Ultimaker/Cura #3739: Using retract in START GCODE causes under extrusion at start of job

Scanjo, 2018-04-27:
**Application Version**

3.2.1

**Platform**

Windows 7

**Printer**

Not relevant

**Steps to Reproduce**

Added code to START GCODE to do a prime line. Set Extruder to zero. Add retract command in preparation to move print head and start job. Printer starts printing without unretracting. I can find no setting in Cura to solve this issue. Moving the print head without a retract leaves a drool line behind and sucks filament from the nozzle as it moves.

**Actual Results**

Printer does not unretract and filament is under extruded for a time.

**Expected results**

The GCODE should have a command to move the extruder to zero before running the job. This would allow users to complete various tasks before the start of the actual print and have a known start point for the extruder when the job starts. It should come before any other job code such as a skirt, brim, or raft.

**Additional Information**

smartavionics, 2018-04-27:
See https://github.com/Ultimaker/CuraEngine/pull/665 . This should be in 3.3.1.

Scanjo, 2018-04-28:
Downloaded and installed 3.3.1. It is indeed included. Sorry for wasting your time.

closed 2018-04-28 by Scanjo

rossgbaker2, 2018-12-06:
I have verified this problem in 3.6.0.  After my start GCODE Cura inserts a retraction before beginning printing the first layer (in this case it is a raft).  It Fails to initially unretract before beginning the print.

;FLAVOR:Marlin
;TIME:15203
;Filament used: 7.28771m
;Layer height: 0.1
;Generated with Cura_SteamEngine 3.6.0
M140 S70
M105
M104 S200 ; manually editted to heat both bed and extruder simultaneously
M105
M190 S70
M109 S200
M82 ;absolute extrusion mode
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops 
G28 Z0 ;move Z to min endstops
M420 L S1
G1 Z0.20 F{travel_speed} ;move the platform down 0.2mm
G92 E0 ;zero the extruded length
G1 Y-3.0 F500.0 ; move out of print volume
G1 X60.0 E9 F500.0 ; start purge
G1 X100 E12.5 F500.0 ; finish purge line
G1 E-1 F500                ; Retract filament by 1mm to reduce string effect
G1 Z15 F5000               ; Raise extruder to safe Z height (1st shearing action)
G1 X5 Y100 Z15 F5000       ; Move to center left of bed (2nd shearing action)
G1 E1.0 F500 : unretract filament
G92 E0 ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...
M900 K0.33
G92 E0   ;;;; **### _Here is where CURA generated code begins_**
G1 F1500 E-1.5 ;  **It does a retract before moving the head**
;LAYER_COUNT:109
;LAYER:-9
M107
G0 F7200 X67.088 Y76.766 Z0.36
;TYPE:SUPPORT-INTERFACE
G1 F1500 E0.04158  ; **this is nowhere near the 1.5 retraction done above.  It may be the 0.1mm extra prime amount**
G1 F1350 X76.904 Y66.965 E1.28725
G1 X77.2 Y66.704 E1.32269
G1 X77.435 Y66.522 E1.34938
G1 X77.672 Y66.362 E1.37506

So the bottom line is that several segments are drawn with no extrusion.

rossgbaker2, 2018-12-07:
I "I retract" my statement above.  This was an error in my start code, not cura!

Jilocasin, 2019-07-13:
I got the exact same behaviour and can't get my head around why.

Your "G1 F1500 E-1.5 ; It does a retract before moving the head" (which is immediately inserted right after the custom start code and the generated code) is pretty much the same I get. It looks like Cura issues a retract before starting the print, but does not un-retract when it arrived at the start position?

May I ask how you resolved your problem?