Ultimaker/Cura #3740: Cura 3.3.1 - pause and resume USB printing does not work

maikbooehme, 2018-04-27:
Hello,

I just had to pause a printing via USB, the temperature for the bed and hotend stayed up, but the print did not resume after pressing the resume button.

please investigate.

best regards
Maik

alekseisasin, 2018-05-02:
HI @maikbooehme, what we can do is to test printing via USB using only Ultimaker printers.

dragonred7, 2018-05-02:
Just had the same problem. It worked fine at the beginning and then about 2 days ago it just won't resume anymore. USB printing too.

Sneuper, 2018-05-07:
Same problem here.
Printing over USB can pause printing but resume does not work.
Cura 3.3.1 on Windows 10.  Printer Anet A6.

Ghostkeeper, 2018-06-07:
We cannot reproduce this problem on the printers we have available at our office. Someone that has a printer that shows the problem will have to debug this.

maikbooehme, 2018-06-07:
Hi,

 

could you let me know, what I need to do to debug it or which data you would need ?

 

best regards

Maik

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

maikbooehme, 2018-06-15:
please let me know, what I need to do to collect the debug information to troubleshoot the issue .

Ghostkeeper, 2018-07-24:
We'd need to know how to make the Anet A6 and the i3 Pro B printers react to new g-code commands after not sending g-code commands for a while. Do they go into a certain mode where they timed out and stopped the print? Does it work if you use the jogging in Cura's Monitor stage to keep the printer 'busy' during the pause? We can't debug this much via the internet.

nallath, 2019-01-07:
Since we never got a reply, this will be closed.

closed 2019-01-07 by nallath