Ultimaker/Cura #3741: Cura 3.3.1 - Monitoring page temperature is not updating during heating phase

maikbooehme, 2018-04-27:
Application Version

3.3.1

Platform

Windows 10

Printer

I3 pro B 

Issue: 
During the pre-heating phase of each print, the temperatur for the extruder and the bed is not update , like it was in 3.2.1

VanessaE, 2018-05-19:
Happens for me also.  Makes it look like the program or printer communication is hung up.

Ghostkeeper, 2018-06-27:
Can confirm. Also reproduced for me.

diegopradogesto, 2018-06-29:
I was wondering if this PR https://github.com/Ultimaker/Cura/pull/3855 may fix this...

PeskyWabbit, 2018-07-04:
Can confirm, happens to me as well. It does start printing when it's up to temp, but theres not telling how close it is to being up to temp or if it's even heating or not.

ianpaschal, 2018-08-16:
We think this is fixed but unfortunately we can't test it since we only have Ultimaker 3s/S5s in the office. I'm going to close this but if anyone can confirm that the bug exists in 3.4.1, I'll reopen it and mark it as deferred.

fieldOfView, 2018-08-16:
UMO+ and UM2(+) are no longer supported then?

ianpaschal, 2018-08-16:
They are but we're unable to recreate them on those printers, therefore, we think it's fixed, but are unable to check if that's also the case on an i3...

ianpaschal, 2018-08-16:
But to be perfectly honest, we have one working UM2 and one half-working UMO in Cura team, so no, they're not really a priority for us.

pentar25, 2018-09-12:
Temperature still isn’t updating on 3.4.1. Seems like it’s hung with no way to tell. Particularly frustrating with two extruders

pentar25, 2018-09-14:
Update, this appears to only happen when connected via USB. when connected via octoprint everything seems to be perfectly in sync.

alekseisasin, 2018-09-28:
Thank @fieldOfView, he fixed this issue. I want to ask somebody to test this fix. For this purpose, I made **not stable** Cura build. It is only for testing purposes.
The download link is available for 7 days.
https://wetransfer.com/downloads/6ecbe5af589a4d32736a156001df5e2a20180928134317/2c53ab5ce1c1ea32d694a7320c87add920180928134317/aedf1a

PR #4369 (2018-10-03): Fix updating temperature while preheating bed or extruder
This PR fixes updates to the print monitor temperatures while preheating.

While preheating the bed/extruder with M190 or M109, the firmware keeps outputting temperature lines, but these do not contain "ok" because no new command was acknowledged.

This PR is against master, but since it fixes a bug and the risk of regressions is low, it might be considered for cherry-picking to 3.5

Fixes #3741
commit 6bf91d2b (2018-10-03): Fix updating temperature while preheating bed or extruder

While preheating the bed/extruder with M190 or M109, the firmware keeps outputting temperature lines, but these do not contain "ok" because no new command was acknowledged.

Fixes #3741
closed 2018-10-03 by alekseisasin