Ultimaker/Cura #3742: Where is macos interface ?

mikcura, 2018-04-27:
MacOS 10.12.6 and 10.13.4  - Cura 3.3.1 

All file requester are "unix like"

Is it a bug or a new feature ?

[image]

Before (3.2.1)
[image]

fieldOfView, 2018-04-27:
It is neither a bug or a feature; it is a workaround for a bug between Qt and OSX that caused double extensions not to work (eg .curaproject.3mf)

ianpaschal, 2018-04-29:
@fieldOfView is correct. I know it's ugly looking but there was no other option. 😢

closed 2018-04-29 by ianpaschal