Ultimaker/Cura #3744: [3.3.1] custom materials *eventually* show up after several launches and a reboot

WardF, 2018-04-27:
**Application Version**
3.3.1

**Platform**
OSX

**Printer**
N/A

**Additional Information**

Upgrading from 3.2.x, my custom materials did not show up in the `material` drop-down.  Explicitly importing them resulted in a `success` dialog, but they did not show up in the list.  Restarting Cura several times did not solve the issue.  Shutting down the computer at night, starting it the next morning, now all my custom materials (including the duplicates I tried to import several times) are listed in the `Material` drop-down.  Don't know if the reboot had an effect or if it was a more general timing issue. 

I believe I observed a similar issue when updating to the latest 3.2.x, but I chalked it up to a glitch at the time and did not make any formal bug report or record any details.

alekseisasin, 2018-05-02:
Hi @WardF , the issue might be related to the case when cura was not closed good or OS system did give access to the material files. It is assumption. If you can share your old material files and current then probably we can check some cases.
Does it work fine for you now after reboot?

WardF, 2018-05-02:
After rebooting, the materials show up every time I launch Cura.  It just took a while for them to show up.  I cannot say for certain if rebooting was part of the fix, only that the materials did not show up despite restarting Cura several times (using the `Quit` function from the menu), but did show up the next day after restarting the laptop.

I'm attaching the materials files:

* [materials.zip](https://github.com/Ultimaker/Cura/files/1968517/materials.zip)

Thanks!

jackha, 2018-05-09:
Hi @WardF . The restarting sound really weird. Is it possible that you have multiple instances of Cura open? That can explain...

btw. Do all the materials show up or are there still some missing?

WardF, 2018-05-09:
@jackha I am fairly certain that I don't have multiple instances of Cura running at once, and I don't install parallel versions of the app (I'm on OSX).  I know there is a "cura engine", and I'm unfamiliar with the architecture of cura; is it possible that the engine is running in the background after the UI app has closed?  That sounds plausible, but you would know better than I if that's how it works :).

When the materials appeared, they all appeared and none were missing.  Along those lines, the materials were present in `~/Library/Application Support/cura/3.3/materials/`; they just weren't being displayed in the dropdown until the system restart.

jackha, 2018-05-14:
A little explanation regarding Cura and Cura Engine. Cura Engine is the part that actually does the slicing and it returns g-code to Cura. All the material stuff is handled in Cura, in Cura Engine they're just settings. It's impossible that Cura Engine is still running when Cura is closed.

I'm really puzzled now, as I have no clue how this can happen. See if somebody else has it as well or knows what's happening...

WardF, 2018-05-14:
When the next Cura release happens, is there anything I can do *prior* to upgrading that may provide additional information if this happens again? It's a non-issue on my end now that the custom materials are visible, but it feels like if it occurs again there will be a good chance to snag extra diagnostic info, if you want.

jackha, 2018-05-15:
Thanks in advance! You could first make a backup of your preferences and settings, before starting the new Cura. This way you can (hopefully) reproduce any weirdness and share it with us ^^.

Location of preferences and settings: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

Ghostkeeper, 2018-06-13:
May be worth a try with the 3.4 beta being out now.

WardF, 2018-06-13:
@Ghostkeeper @jackha Thanks for pinging me, I hadn't noticed the 3.4 beta available.  I am in fact seeing the same issue; none of the custom materials are showing up in 3.4 despite being in the `$HOME/Library/Application Support/Cura/3.4/materials` folder.  What information can I send you that would be of use?

jackha, 2018-06-14:
Hmm are they always absent or do they show up at some time?
If it's always absent, you can send the 3.4 config folder and hopefully we'll be able to reproduce it. For only the first time it may be more difficult to reproduce.

WardF, 2018-06-14:
@jackha They are thus far consistently absent.  Previous experience tells me they will eventually show up but as of now they aren't.  I'm attaching the 3.4 config folder, I'm curious if anything pops out at you. Thanks!

* [3.4.zip](https://github.com/Ultimaker/Cura/files/2103610/3.4.zip)

Ghostkeeper, 2018-07-24:
Indeed these material files don't seem to load. Looking at the files though it appears that someone has been messing with them using a different editor. There's a couple of differences that cause the files to not load.

The namespace is wrong. Your material files define a new namespace `ns0` and defines every tag in that namespace:

    <ns0:fdmmaterial xmlns:ns0="http://www.ultimaker.com/material" version="1.3">

Cura doesn't know of the `ns0` namespace and thus ignores those tags. A proper tag generated by Cura looks like this:

    <fdmmaterial version="1.3" xmlns="http://www.ultimaker.com/material" xmlns:cura="http://www.ultimaker.com/cura">

Another thing though that's also wrong is the material diameter. The printer that you had open in the folder that you posted needs a filament diameter of approximately 3mm. These materials all have 1.75mm.

After changing these two things I end up with the following file:

    <fdmmaterial xmlns="http://www.ultimaker.com/material" version="1.3">
      <metadata>
        <name>
          <brand>Inland</brand>
          <material>PLA</material>
          <color>Light Green</color>
          <label>Light Green PLA</label>
        </name>
        <description />
        <version>2</version>
        <color_code>#33ff66</color_code>
        <GUID>506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9</GUID>
        <adhesion_info />
      </metadata>
      <properties>
        <diameter>2.85</diameter>
        <density>1.24</density>
      </properties>
      <settings>
        <setting key="print temperature">220.0</setting>
        <setting key="print cooling">0.0</setting>
        <setting key="standby temperature">0.0</setting>
        <setting key="heated bed temperature">55.0</setting>
        <setting key="retraction amount">6.5</setting>
        <setting key="retraction speed">25.0</setting>
        <machine>
          <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2 Go" />
          <setting key="print temperature">200</setting>
          <setting key="standby temperature">150</setting>
          <setting key="processing temperature graph">
                    </setting>
          <setting key="heated bed temperature">60</setting>
        </machine>
        <machine>
          <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2+" />
          <setting key="print temperature">200</setting>
          <setting key="standby temperature">175</setting>
          <setting key="heated bed temperature">60</setting>
          <hotend id="0.6 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
          <hotend id="0.8 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
          <hotend id="0.4 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
          <hotend id="0.25 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
        </machine>
        <machine>
          <machine_identifier manufacturer="Ultimaker" product="Ultimaker 3 Extended" />
          <setting key="print temperature">200</setting>
          <setting key="print cooling">100</setting>
          <setting key="standby temperature">175</setting>
[... 40 lines cut ...]
          <hotend id="0.4 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
          <hotend id="0.6 mm">
            <setting key="hardware compatible">yes</setting>
          </hotend>
        </machine>
        <machine>
          <machine_identifier manufacturer="Ultimaker" product="Ultimaker Original" />
          <setting key="print temperature">200</setting>
          <setting key="standby temperature">150</setting>
          <setting key="heated bed temperature">60</setting>
        </machine>
        <machine>
          <machine_identifier manufacturer="Ultimaker" product="Ultimaker 2 Extended" />
          <setting key="print temperature">200</setting>
          <setting key="standby temperature">150</setting>
          <setting key="processing temperature graph">
                    </setting>
          <setting key="heated bed temperature">60</setting>
        </machine>
      </settings>
    </fdmmaterial>

That material loads properly.

closed 2018-07-24 by Ghostkeeper

WardF, 2018-07-24:
Interesting, thank you for following up and sharing your findings.