Ultimaker/Cura #3746: [3.3.1] Line Width reporting error in Settings>Profile>Manage Profiles ...

Bobb1, 2018-04-28:
**Application Version**

3.3.1
**Platform**

Windows 10
**Printer**

Custom - MakerSelect v2.1
**Steps to Reproduce**

Click on Settings, then Profile, then Manage Profiles ... and then click on current profile
**Actual Results**

The Line Width in the current profile is set to 0.4 mm, however, the value reported on Manage Profiles... is different.  (Even though it reports that they are the same.
**Expected results**

Some report the same value as the default line width in the profile
**Additional Information**

I've attached a screenshot of what I'm seeing.
[image]

diegopradogesto, 2018-05-08:
Can you send us this profile? Preferences>Profiles>Export. Or even better, you can send us a project file (even empty) File>Save Project.

jackha, 2018-07-12:
Hi @Bobb1 
How did you create those profiles? I indeed see the 0.42 and the 0.4 values. However:
If I create a profile myself by changing the line_width to 0.42 and save it (in Cura 3.4.1) , then the layer width is 0.42 in only the extruder, not in the global settings.

Bobb1, 2018-07-12:
When I first created the profile, I just entered the values into the custom fields.  The line width setting initially may have been set .42 mm, however, once I changed it to .40 mm and updated the Profile, doesn't that reset the Global settings for the profile?  I may be misunderstanding the Update Profile with current settings function.

jackha, 2018-07-12:
Did you use an older version of Cura before? I suspect that this arose from a version upgrade, which happens every time when you start a new version of Cura.
If I change and update my profile, the value is still in the extruder and it stays there.

Ghostkeeper, 2018-08-27:
> The line width setting initially may have been set .42 mm, however, once I changed it to .40 mm and updated the Profile, doesn't that reset the Global settings for the profile? I may be misunderstanding the Update Profile with current settings function.

"Update profile with current settings" moves all unsaved changes into the current profile. However this change you made is a per-extruder setting so it gets set in the profile for just that extruder. For you this doesn't matter much since you have only one extruder, but it is something that Cura must support for multi-extruder printers.

You shouldn't be able to get the Line Width setting to be overridden in the global profile. That's why I'd like to know if this can still be reproduced by adding a new printer and creating a profile there. Jackha and I are not able to reproduce this bug without loading in your profile.

nallath, 2019-01-07:
Since we never got a response, this will be closed.

closed 2019-01-07 by nallath