Ultimaker/Cura #3747: [3.3.1] Support Blocker not visible, shows massive model size

stevenfayers, 2018-04-29:
[cura.zip](https://github.com/Ultimaker/Cura/files/1958531/cura.zip)

Thank you for using Cura!
-->

**Application Version**
3.3.1

**Platform**
Windows 10 64bit

**Printer**
Creality CR10

**Steps to Reproduce**

load STL
add support blocker

**Actual Results**
Model dimensions after clicking 11703.8 x 9317.9 x 5567.0mm Model disappears. Support blocker not visible.
[image]
[image]
[base.zip](https://github.com/Ultimaker/Cura/files/1958527/base.zip)

**Expected results**
Model dimensions before 74.9 x 64.9 x 179.9mm 

**Additional Information**

diegopradogesto, 2018-05-08:
Sorry, I can't reproduce it. I have also taken a look at the logs and I just see some error when writing a file but nothing related with the SupportEraser feature. 
Can you please attach the full data folder (not just the logs and config file) and the project file (not just the STL). That will help us verifying what the problem is.

stevenfayers, 2018-05-13:
Here are the requested files:

[CCR10_base.curaproject.3mf.curaproject.zip](https://github.com/Ultimaker/Cura/files/1998384/CCR10_base.curaproject.3mf.curaproject.zip)
[3.3.zip](https://github.com/Ultimaker/Cura/files/1998385/3.3.zip)

fieldOfView, 2018-05-14:
It would also be helpful if you could save a project after adding a support blocker.

fieldOfView, 2018-05-14:
The issue seems to be with the shader that determines the depth of the scene where you click; Because your GPU does not support OpenGL 4.1, the shader falls back to a version that has not been sufficiently tested. I'll try to fix it.

fieldOfView, 2018-05-14:
Should be fixed by this: https://github.com/Ultimaker/Cura/pull/3813

stevenfayers, 2018-05-14:
Thanks.  Guess got don't need the extra project file now. 
Cheers
Steven