Ultimaker/Cura #3749: loss of detail in rendering of layer view

spattinson, 2018-04-29:
**Application Version**
3.3.1
**Platform**
Ubuntu
nvidia-387.34 display driver

**Printer**
geeetech i3 pro b

**Steps to Reproduce**
slice a model and select layer view and zoom in

**Actual Results**
detail lost, dotted lines not rendered that show individual lines printed
**Expected results**
previous versions you could see actual lines, for example how many lines used to print shells etc
It seems that the dotted lines are not being rendered that show individual printed lines
in the screenshot the left image is 3.2.0 and the right image is 3.3.1
3.3 did not exhibit this issue
[image]

**Additional Information**
screenshots

diegopradogesto, 2018-05-01:
You are seeing the layer view in compatibility mode and that was not changed in the latest version, as far as I know. 
Have you tried disabling in 3.3.1 the "Only show top layers" so you have the exactly same configuration in the layer view?

spattinson, 2018-05-02:
The two options from the layer view "only show top layers" and the option "5 detailed layers" have now disappeared from that menu so I cant test that, IDK why or how that happened. The view has also returned to what I expect - ie I can see all the lines and little areas where there would be coasting etc. So my problem is resolved,
For reference I have had issues with settings migration on the last few upgrades, losing them mainly or corruption that hid printer configuration. The settings for 3.3.1 were all cleared and entered by hand in this instance, Whether that has anything to do with my issue?
  see screenshot of expected rendering of layers and missing layer options mentioned above:
[image]

diegopradogesto, 2018-05-02:
Having those options is because you were running Cura with the "Compatibility mode". This mode is automatically selected if your graphic card does not support a specific version of OpenGL (Maybe you are now trying in a different computer or different OS?). You can also enable this mode in the preferences, but I recommend use the normal mode if possible.
[image]

> For reference I have had issues with settings migration on the last few upgrades, losing them mainly or corruption that hid printer configuration. The settings for 3.3.1 were all cleared and entered by hand in this instance, Whether that has anything to do with my issue?
I don't know which files you lose, but the Version Upgrades are always tricky. We do a lot of tests to see if they work properly (of course) but sometimes weird things may happen. In this case, if you have a clear scenario that we can reproduce, just feel free to file another issue here and we can help you.

Since I see that your issue seems to be fixed, I will close it. If you think it still needs more review, just reopen it.

closed 2018-05-02 by diegopradogesto

jedikalimero, 2019-09-10:
I'm experiencing a similar issue in my laptop but not in my desktop PC. Both are running Windows 10 x64 and the same version of Cura 4.2.1.
I don't have the compatibility mode active. I have tried with it active and inactive, but there is no difference.
My cura displays the options "Only show top layers" and "Show 5 detailed layers on top" in the laptop but not in the desktop PC.
So, with my Cura in the laptop I can only see either 1 layer or 5 layers, nothing more. Also, the extruded material in the layers is represented as flat poligons with no volume instead of the typical shaded and volumetric ribbons shown in the desktop PC.
My laptop computer is a Sony Vaio VPCCB with an Intel Core i5 2450M processor and 8GB RAM. It has an Intel HD Graphics 3000 and an AMD Radeon 6700M. It can run properly the latest version of 3DStudio Max so it seems it should run Cura without any problem.
Only thing I can think of is Cura maybe using the Intel Graphics instead of the Radeon.

[image]

[image]

Ghostkeeper, 2019-09-11:
Could you do something for me then?

Could you:
1. Open up a command line window (Ctrl+R and type `cmd`).
2. Type `cd "C:\Program Files\Ultimaker Cura 4.2"` to move to your Cura installation.
3. Type `Cura.exe` to run Cura from a terminal. This should give slightly more output than the normal log.

I'm expecting a message to appear at the very beginning such as

    [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [105]: Trying OpenGL context 4.1...
    [MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [115]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile

What are you seeing?