Ultimaker/Cura #3750: Cura 3.3 and 3.3.1 fail to start (OSX 10.13.4)

jimhigson, 2018-04-29:
3.2 was starting fine.

After upgrading to either 3.3 or 3.3.1 always get this message:

[image]

[image]

It looks like 3.3 can't read the 3.2 config for materials. The most interesting bit seems to be this:

```
packages/cura/Machines/Models/BrandMaterialsModel.py", line 124, in 
AttributeError: 'NoneType' object has no attribute 'upper'
```

I can send my config files in if that is helpful.

Error traceback:
----------------

Traceback (most recent call last):
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/Machines/Models/BrandMaterialsModel.py", line 58, in __init__
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/Machines/Models/BrandMaterialsModel.py", line 124, in _update
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/Machines/Models/BrandMaterialsModel.py", line 124, in <lambda>
AttributeError: 'NoneType' object has no attribute 'upper'

Full log:
--------

Thread 0x0000700006209000 (most recent call first):
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700005d06000 (most recent call first):
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700005780000 (most recent call first):
  File "/private/var/folders/35/pfk85v5x1td46k__w_657mfw0000gn/T/AppTranslocation/FA77FEE5-5356-4C81-AFA1-A94689E0E1DE/d/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 73 in _updateThread
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070000527d000 (most recent call first):
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700004d7a000 (most recent call first):
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700004877000 (most recent call first):
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 293 in wait
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 116 in run
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/threading.py", line 882 in _bootstrap
[... 75 lines cut ...]
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 364 in _logInfoWidget
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 214 in _createDialog
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 82 in __init__
  File "/Users/ultimaker/build/3.3/build/inst/bin/cura_app.py", line 101 in exceptHook
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Qt/Bindings/ControllerProxy.py", line 98 in _onActiveStageChanged
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315 in __performEmit
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 212 in emit
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/UM/Controller.py", line 162 in setActiveStage
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 768 in runWithGUI
  File "/Users/ultimaker/build/3.3/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 687 in run
  File "/Users/ultimaker/build/3.3/build/inst/bin/cura_app.py", line 141 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/Users/ultimaker/build/env/3.3/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

modem7, 2018-04-29:
Looks like it's the same issue as #3734

Ellecross, 2018-04-30:
@modem7 perhaps it is, but checking both of your logs seem to have a different message. I asked for your config folder in #3734, when you can. :) 

@jimhigson haloo. Can you please attach your 3.2 (roaming) configuration folder here please? I'll try to replicate your issue. Thanks!

jimhigson, 2018-05-01:
Yes, I can do that. What is the location of those files under OSX?

diegopradogesto, 2018-05-07:
@jimhigson you can locate the files in `$USER/Library/Application Support/cura/3.3`. Just package this folder into a zip file and attach it to this ticket.

jimhigson, 2018-05-21:
Have attached my 3.2 and 3.3 settings directories, zipped up.

[3.2.zip](https://github.com/Ultimaker/Cura/files/2024051/3.2.zip)
[3.3.zip](https://github.com/Ultimaker/Cura/files/2024052/3.3.zip)

diegopradogesto, 2018-05-23:
The problem seems to be one of your custom materials. Loading it in 3.2 I can see this:
[image]
Did you manually modify it? The issue is that there is not label in the xml file, but instead `<label />`.
In any case, Cura should not crash, so I will mark it as a bug.

Thanks for the info.

diegopradogesto, 2018-05-23:
I already did a quick fix for this in b8417bcccbde323159af5dcdf6d5bb29ca70bbf3.
This should not crash in the next release.

closed 2018-05-23 by diegopradogesto

Ghostkeeper, 2018-06-29:
@jimhigson Could you try this with the 3.4 stable release?