Ultimaker/Cura #3751: Makerbot flavor custom printer is using some wrong commands at the beginning of gcode

JDRitchey, 2018-04-30:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Makerbot clone Klic N Print

**Steps to Reproduce**
1) create custom Makerbot printer profile, including custom start/end gcode
2) create simple slice of stl file
3) save as .gcode

**Actual Results**
gcode is produced, but it has an incorrect command sequence, that runs before the custom start gcode,
and forces printer to wait until heated bed is at maximum reachable temperature, and then prints the skirt / perimeters at that temperature, before returning to the correct temperature.

    ;FLAVOR:Makerbot
       #@@;TIME:2601
    ;Generated with Cura_SteamEngine   ##3.3.1
    T0
    M190 S90
    M104 S230
    M109 S230
    M82 ;absolute extrusion mode

**Expected results**
correct gcode sequence is produced, and printer bed only has to heat to specified temperature before it can start printing.  an example of manually corrected gcode below:

    ;FLAVOR:Makerbot
    ;TIME:2601
    ;Generated with Cura_SteamEngine 3.3.1
    T0
    M109 S90 (changed M190 to M109, because it is not supported on my)
    M104 S230
    (removed M109 S230, as it will cause the heated bed to go to its max temperature)
    M82 ;absolute extrusion mode

**Additional Information**

In Makerbot flavors, the M109 command is for extruder, and will not do what is intended.
additionally, Makerbot does not necessarily support M190 command, so it should not be used.

sources:
http://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature
http://reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait (label misleading)
http://reprap.org/wiki/G-code#M190:_Wait_for_bed_temperature_to_reach_target_temp

I've also mentioned this in this comment:
https://github.com/Ultimaker/Cura/issues/3443#issuecomment-385306329

Ghostkeeper, 2018-05-25:
If `M190` is not supported, perhaps we can output this instead:

    M140 S90
    M190 S90

This will first set the bed temperature to 90 degrees (`M140`) which is always supported for Makerbot according to the RepRap wiki. Then it sets it to 90 degrees again and waits (`M190`). For some printers this last command is supported and then it behaves correctly. For other printers this last command is not supported and then it will start printing on a cold bed, but at least the bed will try to heat up during the print then. It's probably too late, but better than nothing.

> In Makerbot flavors, the M109 command is for extruder, and will not do what is intended.

Ah you meant that it is for build plate, not for extruder. We have a serious problem here then though. Is there no command to wait until the target temperature is reached? If we just change all `M109 S230` to `M104 S230` it'll start the print at room temperature and won't even correctly print until a minute or so when the nozzle finally reached its target temperature. And then it'll miss the entire first layer.

JDRitchey, 2018-05-26:
You're right, I meant that M109 is for build plate temperature, and not for extruder.

To get my prints to work with the M109 and M104 command, I'm just having it use M116 in the custom start gcode section.

I suppose there wouldn't be any harm in having that be a part of the Cura generated code, for backward compatibility.

Then it would be:
```
M109 S90 (set bed temp target)
M104 S230 (set extruder temp target)
M116 (wait for temperatures)
```

jackha, 2018-07-03:
We discussed it and we're not going to make this as we don't support the printer or the printer type. The community is of course welcome to contribute.

Ghostkeeper, 2018-08-15:
This has been fixed here: https://github.com/Ultimaker/CuraEngine/commit/5431d7346fa2b0bced1ae8ebc280e6941d264562
The fix will be included with the next Cura release.

closed 2018-08-15 by Ghostkeeper

JDRitchey, 2018-10-15:
@Ghostkeeper This partially fixes it.  In the latest version (updated today) its using M190 to try to heat the bed, and I believe it should be using M109.  The gcode generated for my prints ends up not enabling the heated bed until after printing has started, so first layer may have problems sticking.

Ghostkeeper, 2018-12-19:
I think that's a matter of the start g-code you're using, JDRitchey?

JDRitchey, 2018-12-19:
I don't believe so.  There isn't any M190 command in my start g-code.

Also, I had even more trouble when I tried to do a dual material print.
For whatever reason, M190 gets used every time it switches materials, so I ended up having to find/replace all M190 with M109 to get it to print right.

Here's my start g-code:
```
(**** Start.gcode ****)
M103 (RPM off)
M73 P0 (enable build progress)
G21 (set units to mm)
G90 (set positioning to absolute)
(**** begin homing ****)
G162 X Y F2500 (home XY axes maximum
G161 Z F1100 (home Z axis minimum)
G92 Z-5 (set Z to -5)
G1 Z0.0 (move Z to "0")
G161 Z F100 (home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
(**** end homing ****)
G1 X-110.5 Y-74 Z150 F3300.0 (move to waiting position)
G130 X0 Y0 A0 B0 (Lower stepper Vrefs while heating)
M116 (wait for all temps to reach goal)
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
M108 R3.0 T0
G0 X-110.5 Y-74 (Position Nozzle)
G0 Z0.6      (Position Height)
M108 R4.0    (Set Extruder Speed)
M101         (Start Extruder)
G4 P1500     (Create Anchor)
(**** end of Start.gcode ****)
```

there is a correct use of M109 in the end g-code:
```
(******* End.gcode*******)
M73 P100 ( End  build progress )
G0 Z150 ( Send Z axis to bottom of machine )
M18 ( Disable steppers )
M109 S0 T0 ( Cool down the build platform )
M104 S0 T0 ( Cool down the Right Extruder )
M104 S0 T1 ( Cool down the Left Extruder )
G162 X Y F2500 ( Home XY endstops )
M18 ( Disable stepper motors )
M70 P5 ( We <3 Making Things!)
M72 P1  ( Play Ta-Da song )
(*********end End.gcode*******)
```