Ultimaker/Cura #3753: Cura 3.3.1 wrong machine prefix

cimba007, 2018-04-30:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10 x64

**Printer**

**Steps to Reproduce**

**Actual Results**
Exported stl files have a "defaultish" wrong Prefix CFDPM

**Expected results**
Exported stl files have a prefix according to the printer name

**Additional Information**
[image]

Ellecross, 2018-04-30:
Ahh, the project file name now uses the printer type instead of the printer name. I don't think this is a bug @cimba007. The behavior from 3.2.1 is different as this one is the intended one now. :/ 

For Cura devs, this was CURA-4891. The change came right after we released 3.2.1

cimba007, 2018-04-30:
The new behavior is .. well .. not good. Now you can't distinguish between multiple printers of the same type .. and in addition to that CFDMP takes up 5 characters instead of only 1-2.

My anet a8 has a "small" lcd and I now can't read the full filename.

ianpaschal, 2018-05-01:
@Ellecross is right. The rationale was that it was always meant to show what type of printer that code was intended for. For example my printer was named Bernard so everything got prefixed with B which is sort of illogical when in fact the gcode was meant for any UM3.

Unfortunately I think is one of those situations where no matter which way it goes, half our users will feel like it's the "wrong" choice.

I suppose ideally there would be a way to set the printer prefix as well as the name but that's adding another feature and I'm not sure it warrants the time involved in making it or the visual clutter it adds to the UI.

But what you're seeing is definitely the code working as intended, so I'm closing the issue for now (it can still be commented on).

fieldOfView, 2018-05-01:
For what its worth, I am with @cimba007 that this is not a good change, especially for the Custom FDM Printer crowd.

To mitigate the issue, the prefix can be turned off entirely in the General preferences.

ianpaschal, 2018-05-01:
This is true. The inconsistency is objectively wrong.

ianpaschal, 2018-05-01:
There's two fixes here, the easy one is just to fix the text for "should a prefix be added," to read "machine type" instead.

The slightly-more-work-but-probably-pretty-useful one would be a second box in the very large rename modal to enter a custom prefix which could be whatever the user finds most useful to their work flow. If none is entered we could use the existing behavior of applying the machine type.

What do you think @Appesteijn?

Pheneeny, 2018-05-03:
I think you should definitely allow setting the prefix in the settings. 

From my point of view, using the printer type as the prefix is a hindrance for the majority of users. For one, anyone with more than one printer that doesn't have a pre-defined type in Cura will have all of their G-Code files have the same CFDMP pre-fix, which is a pain. I would also guess that most people that use Cura don't have multiple, inter-changeable printers of the same type. Even with two printers of the same type, they may have different nozzle sizes, different filaments, etc. installed that would make the G-Code printer specific instead of type specific.

Using the printer type would be useful if you have a large number of identical printers at your disposal and you aren't generating the G-Code for a specific printer. That may be the case for a lot of Ultimaker users, but I imagine that is only the use case for a very small number of Cura users.

ianpaschal, 2018-05-04:
I agree but I have tagged our product owner so that he can assess if/when we would put it into a sprint.

AbeFM, 2018-05-13:
+1
The names are so long they take up half the file space. Printing from SD takes careful naming to differentiate on the printer without scrolling horizontally.

Just let me change the prefix somewhere and it'll work out. :-) please expose the used prefix somewhere in the gui.

Ghostkeeper, 2018-05-31:
There is a poll going on about this issue on the Ultimaker forum: https://community.ultimaker.com/topic/23152-ultimaker-cura-prefix/

garvthakkar17, 2018-06-16:
I have a fix What I did Was Headed Over To C:\Program Files\Ultimaker Cura 3.3\resources\definitions And Edited A Prusa I3 Defination (I am using Anet A8) And changed the dimensions in it and I also Renamed Everything And replaced it with Anet A8 And then saved the file as Anet+A8.def.json And moved it into the folder.
After that I checked the Printers section in others and then Anet A8 Was listed There.
But I got a Prefix AA8
So I renamed Everything To Anet 8 Which Changed the prefix to A8 :-)
Problem Solved. Extra Bonus Now You Can Make Your Own Printer Profile and not have to select custom fdm printer. If you want the defination file here is it just copy paste it here C:\Program Files\Ultimaker Cura 3.3\resources\definitions 
[Anet+8.def.zip](https://github.com/Ultimaker/Cura/files/2108531/Anet.8.def.zip)
Thank You :-)

garvthakkar17, 2018-06-17:
@fieldOfView after configuring printers in cura and config file gets auto-generated in the configurations folder.

fieldOfView, 2018-06-17:
Erm, no, you misunderstand. If you put your Anet_A8.def.json into the definitions folder inside the configuration folder, Cura will do its best to update it if you install a new version of Cura. If you install it in the program files folder and then install a new version of Cura (without your file), Cura will crash because it cannot find your definition.

The files that get auto-generated in the configurations folder actually point to the def.json file, If that does not exist, Cura cannot continue with your configured printer.

jackha, 2018-07-11:
We're going to make a template to format your post/prefix/customized print job name
Devs: CURA-5479

varazir, 2018-11-30:
I'm on Cura 3.6.0 still it's only add a prefix based on the machine type and not name. 
Still no way I can custom set the prefix?

AbeFM, 2018-11-30:
At least get the name down to a couple characters. It's FAR from the worst
problem, but it's annoying, setting a shorter default could be an EASY fix?

varazir, 2018-11-30:
Mine is that I have 2 printers with different firmware that I switch between. 
Don't want to Octoprint to overwrite my files.  I have to remember to edit the prefix or remove it every time.

fieldOfView, 2019-12-16:
Since the plugin is available on the Marketplace, I think this issue can be closed.

closed 2019-12-16 by nallath