Ultimaker/Cura #3754: Update profile resets changes

adras, 2018-04-30:
I just created a custom profile. Then I changed Infill Density from 20 to 10 and clicked on "Update profile with current settings/overrides" and the value changed back to 20 again.

This happens with a lot of other settings as well e.g. print temperature.

smartavionics, 2018-05-01:
Please fill out the issue template with details of your system and Cura version otherwise this issue will be ignored for a few days and then deleted - thanks.

adras, 2018-05-01:
Thanks for the hint. I updated the original post

smartavionics, 2018-05-01:
Hi @adras, thanks for updating your issue - that gives people something to work on.

adras, 2018-05-01:
This error is coming from the Uranium project Uranium-master\UM\Settings\ContainerRegistry.py

but I can't figure out the reason for it yet. It seems like my settings are in an _EmptyInstanceContainer for some reason

adras, 2018-05-03:
I think you can remove the Needs Info label now

spacesuitdiver, 2018-05-07:
From what I can tell the software is saving to the wrong extruder or something with all the extruder updates. A simple test to reproduce:

1. Make a custom profile based of coarse
2.  Check "Enable Ironing"
3. Update profile based on changes, ironing ends up unchecking (resetting)

An interesting find, any why I think this is related to the extruder updates:

1. Make a custom profile based of coarse
2.  Check "Enable Ironing"
3. Create new profile based on changes, note that ironing ends up disabling
4. Two .cfg files end up getting created

First is `creality_cr-10_fdmextruder_pla_%232.inst.cfg` which includes `position = 0` and ONLY the change, in this case `ironing_enabled = True`
Second is the new `creality_cr10_pla_%232.inst.cfg` which is just all the old changes, no `ironing_enabled = True` setting.

I think the cloned configurations are missing position = 0 OR old printer configs are busted?

adras, 2018-06-01:
Any updates on this?

adras, 2018-06-01:
Ok, I just deleted the cura folders in AppData local and roaming.

Then I created a new profile based of the 0.2mm profile. I made some changes to various settings and tried to update it. It did NOT work.

Then I duplicated that profile. Made some changes and it WORKED.

Seems like the first profile you create doesn't work, but any duplicated one works.

jackha, 2018-06-06:
Hmmm I see it, at least for the CR-10. I'll discuss in the team.

jackha, 2018-06-07:
We're going to fix it.
Devs: CURA-5453

diegopradogesto, 2018-06-12:
I'm really glad to say that this issue is now fixed and it will be added to the 3.4 release.

closed 2018-06-12 by diegopradogesto