Ultimaker/Cura #3756: Cura crashes when adding printer hostname

Fehlbo, 2018-05-01:
**Application Version**

3.3.1

**Platform**

macOS Sierra 10.12.6

**Printer**

Ultimaker 3+

**Steps to Reproduce**

Adding printers hostname to connect over wireless. Searched for hostname in browser using - http://IP-Adress/api/v1/system/hostname. Copy-paste that into the "add new printer" section from the printer management window. Crash.

**Actual Results**

Whole program crashes instantly.

**Expected results**

It should recognise the printer and connect.

**Additional Information**

This method of connection worked before the latest update.

diegopradogesto, 2018-05-01:
As far as I understand, you mean adding the hostname in this dialog, right?:
[image]
In the dialog for adding a new connection to your UM3, it's suppose to add the IP address, not the hostname.

Fehlbo, 2018-05-01:
Yep, that's the dialog box I'm talking about. Adding the hostname there used to work and it would connect to the printer and it was easy to reconnect each time I used Cura. Now only the IP address works and that changes every day on the Ulti at our work - so I have to go to the printer and find todays IP address and edit it in Cura. Was a lot easier just using the hostname.

fieldOfView, 2018-05-01:
> In the dialog for adding a new connection to your UM3, it's suppose to add the IP address, not the hostname.

Either is acceptable. That's why it says IP address or hostname.

The crash is probably caused by the QNetworkAccessManager in Qt that crashes Macs, see #3580, #2617 and #2622

Ghostkeeper, 2018-05-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/2617.

closed 2018-05-11 by Ghostkeeper

LipuFei, 2018-06-12:
Hi all, we have created a 3.4 BETA build for mac with Qt 5.10 hoping it can fix this issue. Could you try it out see if problem still exists? Thank you all!

Download link: https://we.tl/43i1hChi8m
In case the link expires, let us know so we can upload a new one.

LipuFei, 2018-06-12:
Sorry guys... I only uploaded the linux AppImage, this is the link which contains all installers/AppImage/DMG.
https://we.tl/MLmTYeGuW5

Fehlbo, 2018-06-13:
Thanks Lipu. Just gave this a try. No instant crashing, but also no connection via the hostname. Just "The printer at this address has not yet responded" with no sign of that changing. I can connect fine, as usual through the IP address.

LipuFei, 2018-06-13:
@Fehlbo Thanks! At least upgrading to Qt 5.10 seems to fix that crash. That hostname accessibility can be a different problem.

n1cv, 2019-03-04:
For me, updating the Ultimaker firmware fixed the problem. 
Firmware - from 3.4 to 4.3.
Cura - 3.6.0