Ultimaker/Cura #3759: [3.3.1] A way to control the first layer support grid

brolifen, 2018-05-01:
Sometimes the first layer support grid is just not enough especially when the contact area is very small. I had support structures fail during prints due to very little adhesion surface area of the first support layer. It would be nice to have more control over this layer and specify even a solid first support layer if needed.

Vandrasc, 2018-05-01:
Hi @brolifen, shouldn't support floor settings do the trick? You can even print the support floor with a different extruder for the multiple extruders machines, for that first solid layer. Or, I misunderstood the request...

brolifen, 2018-05-01:
Hi @Vandrasc , the support floor setting only seems to apply on supports starting on top of a 3d model not the ones starting from the bed.

getget1980, 2018-05-07:
I have the same problem and I added a Brim, which helped a lot to keep the support in place.

AbeFM, 2018-05-13:
I to use brim, have suggested brim/raft settings just for support.

I can confirm floor settings don't work.

jackha, 2018-05-28:
We just discussed this in the team and we've decided not to do this.

However with the problems that you mention, you can do several things already to improve several aspects:
- Adhesion type brim is by default used for small adhesion areas
- With modifier meshes you can add or remove support upon existing generated support, or create your own.
- Sometimes a place a thin cylinder as "brim" model somewhere to add extra adhesion where it really needs to stick. You can then place support on top for instance.

closed 2018-05-28 by jackha

AbeFM, 2018-05-28:
I was going to post on here yesterday: how about allowing floor settings to apply to the bed? 

The dense/inflated floor would significantly increase adhesion and require very little extra coding?

AbeFM, 2018-05-30:
[image]

Notice how the top of the image has a nice solid raft-like layer, but the closer half is disparate single line widths directly on the bed?

Here I hid the bottom layer or two to show what it would accomplish:
[image]

By allowing the user to select "apply support floor settings to print bed" or similar, you could allow this.

[image]

jackha, 2018-05-31:
I understand what you mean, but we don't have plans to make this.
In the meanwhile, you can also turn on raft or brim, that should help the support stick.

AbeFM, 2018-06-08:
In cases where the support is pretty small? Well, no one can pretend you haven't heard our ideas - increasing contact area of the supports with single or dual supports is a highly requested feature.

I've tried to suggest some ways that would minimize load on the programming team.

I hope some day if the team decides to work on supports we see some improvement here. Thanks for the consideration nonetheless.

Festivejelly, 2018-10-09:
Just FYI this now seems to be available. Initial layer support line distance just change it to something like 0.1 and you'll get a nice bed interface before support lines.

AbeFM, 2018-10-09:
Yes! It's good!

I've already noticed a small issue - it's easy to set, for instance, your supports at 1.2 mm spacing, and the floor at 1.3, and end up with terrible supports.

It seems it would be better to express it as an integer multiple - 2x or 3x, etc - although high fills, like you mentioned, are better. But I tried one last night at 2x and it was useful!

https://community.ultimaker.com/topic/24741-initial-layer-support-line-distance/

Ghostkeeper, 2018-12-13:
> I've already noticed a small issue - it's easy to set, for instance, your supports at 1.2 mm spacing, and the floor at 1.3, and end up with terrible supports.

Good point.

We would normally set a warning value for this, but our warning value system is currently limited to a minimum and maximum value between which no warning will be given. This means that we can't give a warning for when a setting is not an exact multiple of a certain other value. I'm advocating for a change of the warning and error values to pure formulas, so you could turn the warning formula for this setting into:

    support_initial_layer_line_distance % support_line_distance != 0

And most other settings would turn into something like:

    min_warning < value < max_warning

AbeFM, 2018-12-13:
That would be reasonable, for catching errors in general. And perhaps people would find it limiting to have just a density factor.

But I suggest it's still the better way to handle this setting: 1, 2, 3 or 4 times support density - just make it a factor instead of a spacing. It's like those nutty Europeans with the whole "32 liters per 100 km" thing.  :-P

AbeFM, 2019-04-04:
Other settings are corrected to be multiples of other variables (such as layer height).... I don't know what's going on behind the scenes, but it would seem that this could be enforced.

marekruzicka, 2019-07-10:
> 
> 
> Just FYI this now seems to be available. Initial layer support line distance just change it to something like 0.1 and you'll get a nice bed interface before support lines.

Unfortunately it only seems so... it kind of does it.

If you are looking for a **"solid initial support layer"**, one would expect to set 
`Initial Layer Support Line Distance = 0` which would ideally consider it to be a **"solid layer"** and create a very simple and fast print pattern (maybe it could consider it a `100% infill layer` and apply infill settings).
Since setting it to 0 actually **disables** initial layer, you have to do as you suggested something like 0.1.

Look at the print path it generates... **it's a mess.**

I can't stop wondering how difficult would it be to implement, considering we have
`Support Wall Line Count > 0` -> defines the area
`Initial Layer Support Line Distance = 0` -> converts it to a _"standard solid area"_ or  _"100% infill area"_ for print path calculation.

This issue/suggestion/request was closed just about year ago... maybe it is time to reconsider?

Ghostkeeper, 2019-07-11:
> If you are looking for a "solid initial support layer", one would expect to set
Initial Layer Support Line Distance = 0 which would ideally consider it to be a "solid layer" and create a very simple and fast print pattern (maybe it could consider it a 100% infill layer and apply infill settings).

Just like the normal Support Line Distance setting, the setting simply indicates the distance between 2 centrelines. If you have a line width of 0.4mm and a distance of 0.4mm, it should result in 100% infill. All of the line distance settings in Cura work this way. If you'd set it to 0.1mm you'd get 400% infill. Not practical for many applications but possible nonetheless. Setting it to 0mm is a special case here because infinity% is not possible. We use that as a way to indicate 0% infill. I don't think we should change that for this setting.

marekruzicka, 2019-07-11:
Okay... I did little more testing and you are kind of right. It comes down to _what is exactly is the goal here_.

It will produce a solid layer all the time, **but the way it is done is rather terrible.**

IMHO (close to) ideal situation you can achieve at the moment (talking strictly about 1st layer, since it will pretty much screw up the rest of the supports) is setting high number of **Support Wall Count**. It produces a nice solid layer and does it rather effectively. It works very nice with any other brim settings.

Your suggestion 0.4 width/0.4distance works in very similar way as long as you are using **concentric pattern** for supports (who does?).  Not as nice as doing walls, because after printing brim/walls it jumps into the center and prints in outward pattern (continuing from the brim/walls to center makes more sense for better adhesion). 
It works well for "cross pattern" and for anything else (everything that people are actually using) it produces a messy print paths.

Do you have any particular opinion about **"Initial Layer Support Wall Count"**?
1 new variable, simple condition for 1st layer? (did not study the code yet, so I might be awfully wrong here).

AbeFM, 2019-07-11:
Generally, I like wall count/distance. Both are fine, probably a maximum is what you want, if the two sets hit each other they should be trimmed. I suppose that's obvious. :-)

I feel just a simple "support floor density", expressed as an integer multiplier of the normal support density would capture everything I could want. I want it 2x, or 3x, etc, as dense as the supports. You could make it literally symmetrical to how you do the gradual infill/supports now - but frankly just having it 2-5 times as dense is all I would practically ever need.

Ghostkeeper, 2019-07-12:
Take a look at Support Brim and its Support Brim Line Count setting, @marekruzicka. What you request already exists.

Support floor is a term for support resting on the model right now, so we can't use that for support resting on the build plate.

marekruzicka, 2019-07-12:
> 
> 
> Take a look at Support Brim and its Support Brim Line Count setting, @marekruzicka. What you request already exists.
> 
OMG ... I must have been blind doing all the testing and missing this.

The thing is, I assumed that support brim behaves the same way brim does... expands outwards from the model, so I've never really test it that much.
What is actually happening is that **"regular Brim"** defines the outline of the model (including the supports), and **Support Brim** fills the supports area inward. 

`Enable Support Brim`
`Support Brim Line Count = 100 (high enough number)`

I'll do more testing over the weekend, but after a quick test, this looks like nobrainer.
Thank you @Ghostkeeper.

AbeFM, 2019-07-12:
Huh. I also missed that.

To be fair, loops on the outer wall of the support isn't as useful as increasing the contact density.
[image]
Here's 5 loops, but I don't think it's more stable than having the higher density floor.

CAN SOMEONE PLEASE RESPOND:
Having a non integer multiple on the initial layer adjustment is much, much worse than not having it at all.
[image]

See how the non-integer setting here makes a loose grid for the first layer, THEN the layer above is bridging across the entire support structure.
This isn't meant to be insulting - but that's dumb. Please explain why you would want it?!

**This setting should be an INTEGER multiple**, please explain if you don't agree.

Ghostkeeper, 2019-07-15:
Indeed, I agree. It'd be pretty stupid to set Initial Layer Support Line Distance to something that is not an integer division of Support Line Distance. Almost as stupid as setting your temperature for PLA to 300 degrees.

Our setting warning system currently only allows for a maximum and/or minimum value though. We couldn't give a warning on non-integer divisions of another setting.

AbeFM, 2019-07-15:
Can't you make the input itself a pull-down-select-a-value deal? I can pick
"brim" or "skirt", but not "brim.5". :-)

AbeFM, 2019-08-01:
Am I on to something here? I.E. Will a pull down work, do you need me to keep making more suggestions or is it just a question of timing?

Ghostkeeper, 2019-08-11:
The drop-down's values would need to be dynamically filled then, right? I mean, if the Support Line Distance is 2mm, it'd need to be filled with 0.2, 0.4, 0.6, 0.8, 1.0, etc. but if the Support Line Distance is 2.1mm it'd need to be filled with 0.21, 0.42, 0.63, 0.84, 1.05, etc.

It's not a type of setting that we have right now, though it is feasible in theory. Also needs some thinking from a UX point of view.

AbeFM, 2019-08-12:
Make it populate with:
1x
2x
3x
4x

That's it. Call it support floor density multiplier.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes