Ultimaker/Cura #3761: Parameter tabs by category

Hebign, 2018-05-01:
Just a suggestion for the visual conford and a better approach to the settings ...

Set up tabs grouping the settings by categories rather than just one vertical list.

Coming from Simplify3D and IdeaMaker, i find Cura 3.3.1 really better, but very visually tiring to navigate the settings.

Thank you.

ianpaschal, 2018-05-01:
They are grouped by categories are they not?

Hebign, 2018-05-01:
Yes, but not in the form of tabs per parameter section, which makes, I find, the path of the parameters very rough and lost time to find

Hebign, 2018-05-01:
you know Simplify3D, IdeaMaker ... you see the organization of the parameters? well the same thing ... Each type of parameter has its tab, for example, a tab for shell, another for support, another for filling, etc. ... not complicated to imagine, however, I do not know if is complicated to set up, I'm not developing ...

Hebign, 2018-05-01:
but is this your way to make me understand that Cura is well like that and that nobody will be interested in my idea? who I am sure would please a lot of people! even if I think I'm not the first to think about that ...

fieldOfView, 2018-05-01:
The idea of tabs is not really compatible with a sidebar like Cura uses. @mennoberg once (2.5 years ago) did a sketch of a more tabs-based version, but you can see how that would need more space (and would not make efficient use of the sidebar):
https://github.com/Ultimaker/Cura/issues/194#issuecomment-127529878

So switching to tabs would mean going away from a sidebar, reimplementing a big part of the UI. In the mean time people have been getting used to the sidebar and things like sidebar search and setting visibility profiles, which are not compatible with the concept of tabs. Chances of switching out the UI any time soon are slim to none.

Hebign, 2018-05-01:
I imagined buttons along the separation of the slice view and parameters, vertically, to display each category ... No need for scroll bar, finally, no more than it There are already, or change the format of the settings, just pages, activatable via buttons, like web CSS ...

ianpaschal, 2018-05-01:
It may make some people slightly happy but there's little evidence it would make Cura more usable as it already has many helpful features to locate features (such as collapsing sections, search bar, and presets). @LukeChen-Ultimaker is also already hard at work analyzing how this can be improved.

But I can assure you re-designing the whole UI to use tabs is not going to happen from within the Cura team.

Sorry about that! We do appreciate the suggestions though.

closed 2018-05-01 by ianpaschal

Hebign, 2018-05-01:
I do not know if you have taken the time to visualize my idea, it does not need to resume everything, but, it does not matter, thank you for your work, it's a good soft, I would say it is missing some commands in the control interface of the printer, but it's very good job! Good continuation !