Ultimaker/Cura #3764: Cura 3.3.1 is Crash.

cooljing, 2018-05-02:
**Application Version**
Cura Version 3.3.1

**Platform**
Windows 7 64bit

**Printer**
FreeDMake v1.5(Custom Taiwan printer)

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**
 I updated to Cura 3.3.1 .
But I always run into software crashes.
Everything works fine before I have loaded my old JSON file.
Since I put in the old JSON file, the software could not be opened.

The error message is as follows:
> Traceback (most recent call last):
> File "<frozen importlib._bootstrap>", line 969, in _find_and_load
> File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
> File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
> File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
> File "__startup__.py", line 12, in <module>
> File "<frozen importlib._bootstrap>", line 969, in _find_and_load
> File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
> File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
> File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
> File "Console.py", line 24, in <module>
> File "X:\3.3\build\inst\bin\cura_app.py", line 141, in <module>
> File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 652, in run
> File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Machines\MaterialManager.py", line 107, in initialize
> KeyError: 'GUID' 

The log is as follows:

> Th﻿read 0x0000266c (most recent call first):
> File "X:\3.3\build\inst\l﻿ib\python3.5\site-packages\UM\Backend\Backend.py", line 155 in _storeStderrToLogThread
> File "threading.py", line 862 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in ﻿_bootstrap
> 
> Thread 0x000018c4 (most recent call first):
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Backend\Backend.py", line 147 in _storeOutputToLogThread
> File "threading.py", line 862 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 88﻿2 in _bootstrap
> 
> Thread 0x000019b0 (most recent call first):
> File "C:\Program Files\Ultimaker Cura 3.3\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 66 in _updateThread
> File "threading.py", line 862 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x000021d0 (most r﻿ecent ﻿call first):
> File "threading.py", line 293 in wait
> File "threading.py", line 424 in acquire
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x000003c4 (most recent call first):
> File "threading.py", line 293 in wait
> File "threading.py", line 424 in acquire
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x00001f74 (most recent call first):
> File "threading.py", line 293 in wait
> File "threading.py", line 424 in acquire
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _boots﻿trap
> 
> Thread 0x00001490 (most recent call first):
> File "threading.py", line 293 in wait
> File "threading.py", line 424 in acquire
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
> File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _boo﻿tstrap
> 
> Thread 0x00002214 (most recent call first):
> File "threading.py", line 297 in wait
> File "zeroconf.py", line 1569 in wait
> File "zeroconf.py", line 1153 in run ﻿
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x00002164 (most recent call first):
> File "threading.py", line 297 in wait
> File "zeroconf.py", line 1569 in wait
> File "zeroconf.py", line 1017 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x00001308 (most recent call first):
> File "zeroconf.py", line 946 in run
> File "threading.py", line 914 in _boo﻿tstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x00001408 (most recent call first):
> File "threading.py", line 297 in wait
> File "threading.py", line 549 in wait
> File "C:\Program Files\Ultimaker Cura 3.3\plugins\UM3NetworkPrinting\UM3OutputDevicePlugin.py", line 274 in _handleOnServiceChangedRequests
> File "threading.py", line 862 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Thread 0x00000810 (most recent c﻿all first):
> File "C:\Program Files\Ultimaker Cura 3.3\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
> File "threading.py", line 862 in run
> File "threading.py", line 914 in _bootstrap_inner
> File "threading.py", line 882 in _bootstrap
> 
> Current thread 0x00000ea0 (most recent call first):
> File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 364 in _logInfoWidget
> File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 214 in _createDialog
> File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 82 in __init__
> File "X:\3.3\build\inst\bin\cura_app.py", line 101 in exceptHook

cooljing, 2018-05-02:
My JSON is here.
[resources.zip](https://github.com/Ultimaker/Cura/files/1966572/resources.zip)

ChrisTerBeke, 2018-05-02:
Seems like GUID is missing in one of your material config files?

modem7, 2018-05-02:
Sounds like same issue as #3734 - I'm also missing guid in a material file.

ChrisTerBeke, 2018-05-02:
Closing as it's not an issue with Cura itself (but rather with completely custom files).

modem7, 2018-05-02:
For future reference for anyone who has issues with no guid in their custom material files, please use the following:

https://guidgenerator.com/

fieldOfView, 2018-05-02:
Shouldn't Cura be able to handle this more gracefully? It did in previous versions apparently, so that would make this a regression.

ChrisTerBeke, 2018-05-02:
Fair enought.

cooljing, 2018-05-03:
Thanks everyone~~~
I fixed this problem!!!
I remove the materials JSON file , and it can work.

ChrisTerBeke, 2018-05-16:
Fixed in c13b77d8a19b59c66e20b77cbd1263ec9b5f4a74.

closed 2018-05-16 by ChrisTerBeke

Ghostkeeper, 2018-05-28:
That fix was undone here: https://github.com/Ultimaker/Cura/commit/ddc9da63b85644d62eb1ccb87180b0aaf2c546e2

And fixed in a different way here: https://github.com/Ultimaker/Cura/commit/f5a77071391f37d65f07d22e7bbd967cb706094e