Ultimaker/Cura #3766: 3.3.1 Cannot add custom material

modem7, 2018-05-02:
Version 3.3.1

Windows 7 x64

Wanhao i3 2.1

When trying to add custom materials, an error occurs when changing diameter, regardless what the custom printer is set as.

[image]

[image]
[image]

Also, unable to import material as well.

[image]
[image]

diegopradogesto, 2018-05-03:
If you create a material with 1.75mm diameter but the selected printer is configured to use 2.85mm filament, then the material is filtered out, that's why we show the message. If you switch to a printer that uses 1.75mm filament, then your material will be there.
Be sure your printer has the same setting in Settings>Printers>Machine Settings and then in the Extruder tab, take a look at the compatible material diameter.

modem7, 2018-05-03:
Please see the screenshots in the original post.

It shows the printer extruder size set to 1.75.

diegopradogesto, 2018-05-03:
Mmmmm, ok sorry. I will try to reproduce it.

modem7, 2018-05-03:
Just in case it was values in custom printer:

200x200x180 w/ heated bed. No further changes at this time

diegopradogesto, 2018-05-03:
I used your user data (the one that you posted on the other thread) and I'm able to create new materials and duplicate them.
Can you please try to do it with a clean install (removing your data folder - make a backup just in case). If it still doesn't work, can you please save a project file (File>Save as project) (even without any model in the buildplate) that allow us to get your current configuration.

modem7, 2018-05-04:
Clean install resolved. 

Where would previous printer settings from old versions be stored? I can't seem to find them!

Tried machine_instances but nothing in there for full printer settings

closed 2018-05-04 by modem7

Ghostkeeper, 2018-05-28:
The old versions are in a folder next to it. For you on Windows that would be:

    C:\Users\modem7\AppData\cura
    | - <some log files are here>
    |-3.2
    |  |- <your configuration from the last time you used 3.2>
    |-3.3
    |  |- <your current configuration in 3.3>

RichyT, 2018-12-28:
Have the same issue, Curea 3.6.0 with a custom printer. I would suggest that reinstalling the software is not a sufficient fix (only a work-around) considering this occurred with a clean install.

ttait-vantim, 2019-01-26:
Fresh install on 3.6.0 on new PC and couldn't see any imports either... then I checked the extruder and it was set to 2.85mm for 1.75mm machine. Stranges as this was the same custom JSON for the Craftbot+ I've been using since 3.2.  I changed the extruder daimeter in the GUI and now all is good, but I can't find where that change was stored...