Ultimaker/Cura #3767: scaling or something is major FUBAR

dragon3241, 2018-05-03:
**Application Version**
Cura 3.3

**Platform**
Windows7

**Printer**
Maker Select Plus>

**Steps to Reproduce**
create the gcode, and it looks good but it does not print correct at all

**Actual Results**
printer does not find home, just starts anywhere out in the perimeter of build plate, and runs off the plate even though the build plate is set correctly.  it also seems to be doing some kind of scaling, although scaling is set to 100% and in the view, it looks perfect, but what it tries to print is not even close to reality

**Expected results**
i would expect the printer to try to do what it shows in the program view, slices show correctly, but it does not function

**Additional Information**
tried and retried, rebooted multiple times, nothing is working.  it just keeps doing it wrong.  i can run the printer on test item, and it works right, but when i try to run a new project, it changes the actual size, despite the settings and despite what it shows on the screen.

dragon3241, 2018-05-04:
since i really don't have the right terms to use, lol, i made a short video for anyone to look at and see what it is i am trying to describe.

https://www.youtube.com/watch?v=1qM7W6FkhbU

i will attach one of the files shown in this video.

thanks everyone!!

Well, OK, lol, i guess i won't attach a file.  i can't seem to figure out how to do that with g-code files, it tells me that it is not a supported file type.

hopefully, the video will show someone what it is i am facing.

smartavionics, 2018-05-04:
Just zip it or rename the gcode to a .txt file and attach that, thanks.

fieldOfView, 2018-05-04:
Since there is a 10Mb size limit, zipping would be the better option. Another option would be to use eg dropbox or google cloud etc to get a publicly accessible download link and post that link here.

dragon3241, 2018-05-04:
OK, this site will not accept a zip file if I do it with the built in zip utility 7zip.

So I tried to copy the g-code, which I now have in a word document, onto Google Docs, and it failed, lol, 3 damn times.

So I have 2 word docs, containing 2 zip code listings, one is pretty short, the big one is almost 7 megs, and I am told I can't post a file that big.

Soooooo, guess I will thank you for the try, lol, but I don't seem to have a way to do this, I guess I will try getting in touch with Cura directly, maybe they will help me.

Thanks again though.  :)

ianpaschal, 2018-05-04:
> I guess I will try getting in touch with Cura directly, maybe they will help me.

Are you going to show up at our office? You can get to us directly [here](https://www.google.nl/maps/place/Ultimaker/@51.8805759,5.2630119,17z/data=!3m1!4b1!4m5!3m4!1s0x47c6f5c27d2f2329:0x542ec838c9376533!8m2!3d51.8805759!4d5.2652059).

Just teasing. 😛 But this is the most direct way to reach the Cura developers.

fieldOfView and smartavionics are two of the (or perhaps just 'the') most helpful members of the community, so trust that they're doing their best to help you out.

The truth is they're right. I watched most of your video (skipped around a bit because it's sort of long) but what we really need in order to diagnose this is the gcode. There's a lot of ways you can do this: right click and select either "compress" (macOS) or "Send to" and then select "Compressed (zipped) folder" (on Windows).

Another option if you use dropbox or Google Drive is to upload it there and send a link, or finally, use [WeTransfer](https://wetransfer.com) to freely send big files. You can also send it to 
[image] (either normally by email or with WeTransfer).

From your video though three things stood out to me:

1. Use the layer view to see the actual gcode visually! This is a really good way to inspect the gcode without having to start a print. It's not foolproof but it's pretty darn reliable.
2. It looks like your homing command is missing as the L shape looks like the "bottom" of the 3 wings in terms of size relative to the build plate, and the other two would be created "above" it which in this case is off the build plate. Something like the 0 point of that axis being the center of the build plate instead of the front edge of it. But again, impossible to really verify this without seeing the gcode.
3. Finally, your printer, the Monoprice Maker Select Plus doesn't seem to be one of our include printer definitions. It would help if you save your project as a .curaproject.3mf and upload or email *that* using the above methods because that will include the printer configuration and we can see if something went wrong in that part (I'm assuming since the printer is not included in Cura that it's a custom definition you made).

Hope this helps!

dragon3241, 2018-05-04:
lol, actually, I did send an email to the office asking for support, I got the address from the website.  I confess I did not know that this site, lol, was related to CURA, I just kind of assumed it was a forum board.

as for the g-code files, I did try what I knew how to do, lol, but sadly, the site did not cooperate with me, so while I was VERY grateful to the people who tried to help me, I just kind of figured I was stopped here.   Which is why I sent the email to the office.  

And no, lol, you are clear across the country from me, I won't be dropping into your office,,,,,but if you were closer........   :)

But I will send the two files to your email that you showed in the last post, perhaps that method will work for me????

smartavionics, 2018-05-04:
I took a quick look at the gcode and couldn't see anything obviously wrong but the part is printing around an origin of 0,0 which unless the printer is a delta is going to make it try and print off the bed.

Check the machine settings and look to see if you have origin at centre enabled?

[image]

[image]

ianpaschal, 2018-05-22:
Haven't heard back in ~3 weeks so I'm going to assume that was the issue and it's resolved.

dragon3241, 2018-05-22:
Lol

 

And I had not heard anything more, so I kind of assumed that I was just SOL, hahahahhahahahahahah

 

I did find what I think was the solution however.  It seems that my printer requires using the version 15 CURA ONLY, none of the newer versions will work with it.  In actual fact, it seems that there was an update to mine, because the version now shows a 16.02

 

But it is back to working, so I guess I am good.  Sure wish I could use the newer version though…..

 

Lee

 

 

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

dragon3241, 2018-05-23:
> We had asked if you could share your printer definition or project file because the GCode was fine, and didn't hear back from you.
>
> Ideally we get this worked out because Cura 15 is very old and no longer supported or updated.

Ahhhhhhhhhhhhh,

 

Lol, never got that email.  😊

 

Well, yeah, I would love to be able to use a newer version.  What can I get to you, so that you can tell me how to make it work with a newer version?  I warn you, lol, I am a total newbie at this 3D printing stuff, lol, so be gentle.  Hahahahahhaha

 

I can tell you that it is a Monoprice Maker Select Plus, and that it is a Chinese copy of a Chinese printer, a Wanhao 3P.  I know that the motherboard and build plate are or seem to be, original Wanhoa parts.  At least that is what the silk screening says, lol, lord only knows if it is legit or not???

 

When I got the printer, it came with a ‘version’ of Cura on a memory card, my understanding is that it was supposed to be version 15, but when it opens up on my laptop, it says version 16.02.

 

This ‘printer profile’, what is it, and how do I find it so that I can send it to you?

 

Because for sure, lol, I was unable to use versions 2, 3.3 or 3.5 on my printer.  Maybe the G-code was fine as you say, lol, but I gotta tell you, lol, it wouldn’t print what I wanted, not even close.  So something about the G-code was not interfacing or communicating right at all.

 

But those same files, ran through the 16.02 I have, work OK.

 

So tell me what you need, and how I go about getting it for you, lol, I am happy to comply.  Just don’t know enough to even be very dangerous at this point.  😊

 

Lee

 

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

ianpaschal, 2018-05-23:
If in Cura 3.3.1 you set up your print as expected, then go File, save as project, you will get a .3mf file which contains the models as well as all the configuration details of your printer and the settings you're using.

Go ahead and send that file via WeTransfer to 
[image] and we can take a look at it.

dragon3241, 2018-05-26:
Hi Ian,

 

Just wanted you to know that I have sent the files, plus some ones you did not ask for, lol, via WE TRANSFER, so they are there for the next 7 days.

 

Have a great weekend!!

 

Lee Rast

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

ianpaschal, 2018-05-26:
Thanks Lee, I got them. I have some time at home on Sunday I'll take a look.

ianpaschal, 2018-05-26:
Also since this is still a work in progress, I'll reopen.

dragon3241, 2018-05-26:
Well just don’t spoil your holiday weekend for me!!!  Enjoy your family and friends, celebrate and remember those of us no longer here, and have a BBQ!!!

 

Lol

 

My project isn’t going anywhere for the next three days anyhow.  😊

 

 

Lee

 

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

Ghostkeeper, 2018-06-04:
One thing you could try is to add this line to the end of the Start G-code:

    G28

Second thing you could try is what SmartAvionics mentioned, to check the "Origin at center" checkbox in the same screen.

You can find both of these by clicking on the printer name in the top right of the Cura screen, clicking "Manage Printers...", then "Machine Settings". You'll see the Origin at center checkbox there as well as a big text box where you should add `G28` at the end on a new line.

I don't know if @ianpaschal still has those files and if he remembered to take a look at them?

dragon3241, 2018-06-07:
Hi Ian,

 

Just thought I would touch bases with you, and see if you have had a chance to look at all that stuff I sent you, and figure out how what and why, lol, also, do you have any idea if we will be able to somehow use the newer version of CURA with my printer?

 

Not trying to joggle your elbow, lol, I know I am not the only project you have to work on, lol, but just thought I would drop you a line and see what or if you knew anything yet.

 

Thank you very much for your time and efforts!!!

 

With respect,

 

Lee Rast

 

 

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

 

Reopened #3767.

ianpaschal, 2018-06-21:
Hi Lee, The files you sent are not the ones we were initially trying to diagnose. Makes it very tough to figure out what's going wrong with one print using a project file from something else.

dragon3241, 2018-06-21:
Well, the only files I had on the original failing part, was the one I sent you way back then.

 

I kind of assumed, lol, that sending you the files from this different part, that was having the same issue, would be OK.  Guess I was wrong.

 

I will see about getting the file from the original part, and trying to run it through all the phases I did with the ones I sent you, and see if that helps.  Probably won’t get that done until this evening, but I will get it done and then let you know when it is available.

 

I’m not sure what happened, but back when this all started, the ‘wing’ file was also one I was trying to work with, and I thought, lol, that I had sent that one too, back then.  Perhaps I was mistaken?

 

Lee

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

dragon3241, 2018-06-27:
Hi Ian,

 

Sorry for the delay, I was a bit ill for a few days.

 

OK, here is the file that I originally sent thru github.  I have included 3 versions of it.

 

1 is the obj file I used to get into the CURA program

1 is the resultant gcode from CURA 16.02 that I am currently forced to use on my printer

1 is the same obj file but going through CURA 3.3.1 to produce the gcode file that won’t work properly for me on my printer

 

You should find the same kind of problem with this that I sent you in all those other files, just a different part.  If you decide you want me to, I can also print up both files and send you the video showing the difference between the two gcode.

 

Thanks Ian!!!

 

Lee

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

Ghostkeeper, 2018-06-28:
Attachments don't work when you reply to a notification from Github about a comment being added. Github is kind enough to post your reply on the appropriate issue thread, but won't upload the attachments automatically. Could you send the attachments in some other way, like WeTransfer that you used before?

ianpaschal, 2018-06-28:
I got them.

@dragon3241 I was at a trade show out of the country one week, then sick one week, then working on our research sprint for 2 weeks. Now we're back to business as usual. It's 5 o clock here, but I'll take a look tomorrow morning.

dragon3241, 2018-06-28:
Ahhhhh, lol, sad how it seems that things always come back to doing a routine, lol, like a job.  Hahahahhahahahahahaha

 

Here are those three files.  I thought I sent them to you direct, but didn’t pay attention to the actual address when I sent them the other day.  😊

 

Lee

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

ianpaschal, 2018-07-03:
Hi Lee,

I've been looking at the files and watching your videos.

Regarding the support on the pink part, could that be because fo the "only on build plate" option for support? If that is checked, support will only be generated off of the build plate or raft, but not off of the part, which is why only the lower nub has support under it.

As for the pieces, the models and gcode look fine which suggests to me that either your printer is incorrectly configured in Cura (perhaps something with the "origin at center" option needing to be toggled on or off), or with the firmware on your printer.

Unfortunately these aren't really issues that we as a software team can help you sort out, especially since you're on a 3rd party printer. No one really likes hearing this but the truth is we have only Ultimaker printers at the office so it's very difficult (i.e. impossible) to work on code changes and then test if they're working on a printer we don't have.

Best thing is to experiment with some of those options and see what effect they have. I'd also try printing smaller stuff, like 10 mm cubes, to sort out the hardware configuration before wasting time on a bigger part.

dragon3241, 2018-07-03:
Hi Ian,

 

I had also figured out the support issue being the selection of everywhere vs. off the build plate, so that part is solved.  The other thing with it building NOT in the center, that selection is in fact checked, but for whatever reason, the newer versions of CURA do that kind of weirdness on my printer.  They also skip the steps where it would make sure that the filament is flowing good first, and it does not go to the ’all home’ position first.  What it does do, is start to print and whatever table position the printer happens to be at.  And, it can decide to try and print off the side of the table area as well!!

 

I understand that since this is a third party printer, you are limited in what you can try to fix, lol, I was desperately hoping that you would happen to know WHY these weird things happen, since you guys make the software that my printer demands it work with.  I guess I was hoping that you were going to tell me about some kind of special instructions that were buried in the software version I have to use, that I could somehow copy and paste ‘that’ into the newer versions.   Or something like that.  😊

 

Would there be any use in me sending you the software itself?

 

Talk to you later.

 

Lee

 

 

 

da’  Dragon

 

"Absent Companions"

 

"If you have to ask, you don't need to know."

 

"Lord, make me fast & accurate, let my aim be true and my hand faster than those who would seek to destroy me; grant me victory over my foes and those that wish to do harm to me and mine; let not my last thought be, If I only had my gun; & Lord, if today is truly the day that you call me home, then let me die in a big pile of empty brass."  a gunbunny's prayer

“The seagull flies into the sunset, towards the horizon; although it is far, it doesn’t scare him, for in his mind he sees that he is already there, even before he gets there.”

“Neither enemy faces or the mothers that love them, come to your mind when all you are thinking about is trying to survive.  Debating the merits of war is ridiculous when you are under fire.”

“A soldier doesn’t fight because he hates what’s in front of him.  He fights because he loves what he left behind.”

“When the owl and the witch together are seen, there’s mischief brewing on Halloween!!”

"Overkill is underrated, especially by those of limited imagination"

"Once you pull the pin, Mr. Grenade is no longer your friend."

"All for One, and One for All"

"Kick Theirs, Protect Yours"

"Never say with words, what you could express with a flamethrower."

“The problems we face today are there because the people who work for a living… are outnumbered by people who vote for a living.”

"If you force me to do violence, I shall be so savage and so cruel, and hurt you so badly that the thought of revenge shall never cross your mind." -Machiavelli

ianpaschal, 2018-07-04:
> that selection is in fact checked

Bingo! That's your problem. We're actually discussing renaming it because it confuses a lot of people and is, apparently, very unclear what it does.

It's meant to designate your printer as a "delta style" printer which uses a coordinate system originating from the center. Your printer, like most, has its coordinate system originating in the front left corner like a typical X-Y graph.

So it needs to be unchecked. When it's checked it means that your gcode, which if you centered your object in Cura, will be printing it at approximately x = 1/2 build-plate-width and y = 1/2 build-plate-depth will have 0,0 at the center of your build plate printing your object at x = build-plate-width, or y = build-plate-depth, in other words: dangling off the far edge of the build plate.

The only way to confirm this for sure is to, in Cura, with your project set up with the models loaded and your printer selected, to do File > Save Project. Save it somewhere, put it into a zip file, and send that. When you save as a project, not as gcode, you save the whole set up, including your custom printer definition.

But I think if you uncheck that option and try again you'll make progress in the right direction.

closed 2026-04-09 by GregValiant