Ultimaker/Cura #3768: Cura does not retain the Octoprint API key upon restart.

famewolf, 2018-05-03:
**Application Version**
Cura 3.3.1

**Platform**
Linux Mint 18.3

**Printer**
CR-10 renamed to "Tevo Tornado"

**Steps to Reproduce**
Edit the printer settings and enter the api. Connect and send a stl to "Print to Octoprint"

**Actual Results**
Job Prints as expected.

**Expected results**
After closing and re-opening cura the api key should be saved so other objects can be sent...instead I have to enter the key every time.

**Additional Information**
I'm not sure a log will be helpful in this particular scenario but I can do a print to octoprint then close and reopen if it cannot be reproduced on your end.

famewolf, 2018-05-03:
I created a new "instance" of my octoprint info...same ip address and port..saved it under a new name...using THAT instance the api key was retained.  It was NOT retained in the one that auto appeared in the program when I went to that section.

ChrisTerBeke, 2018-05-03:
@fieldOfView

fieldOfView, 2018-05-03:
Without the cura log there is not much I can do.

famewolf, 2018-05-03:
Well after creating a new "profile" for octoprint both are retaining the api so since I can't recreate I'll close this.

closed 2018-05-03 by famewolf