Ultimaker/Cura #3769: Toolbox bug? Can't select z-offset setting plugin

smartavionics, 2018-05-03:
**Application Version**

master

**Platform**

Linux

**Printer**

**Steps to Reproduce**

Open toolbox, move mouse over various plugin entries

**Actual Results**

Moving the mouse over the z-offset plugin entry doesn't highlight it and the entry cannot be selected.

**Expected results**

You know...

**Additional Information**

Notice how the area under the mouse cursor is not highlighted blue.

[image]

ChrisTerBeke, 2018-05-03:
Confirmed issue.

ianpaschal, 2018-05-04:
Yes. Has something to do with how QT handles grid layouts. :/

It's on our "leftover bugs" list.

ChrisTerBeke, 2018-05-04:
I saw that if you add some padding/spacing on the bottom, it becomes clickable again :P

ianpaschal, 2018-05-04:
Yes that's what I thought was in there... there's some line where I check the remainder of the model length and if it's odd add some extra spacing but for some reason that stopped (ehhh?) working?

ianpaschal, 2018-05-04:
Should be in `ToolboxDownloadsGrid.qml`

diegopradogesto, 2018-05-04:
I made a fix in 2283bb530debdcbe6ac4c544f7920377c1cb01dc. Please try if it works.

fieldOfView, 2018-05-04:
Seems to work, be sure to remove the comment above the fix.

smartavionics, 2018-05-04:
Just tried that and it appears to be OK now, thanks. Closing...