Ultimaker/Cura #3772: Submitting a new model configuration file

twosilly, 2018-05-04:
Hello there! I have my own model configuration file! I want him to appear in the next official installation package of CURA software! What should I do?

ianpaschal, 2018-05-04:
What sort of configuration did you make? A quality definition or a printer definition?

In either case, you can:

1. Fork this repository
2. Add your definitions to the right folder (`./resources/qualities/`, or whatever)
3. Make a pull request
4. Wait for us to review your contribution
5. Address any changes that need to be made
6. See it merged!

If you're unsure how to do any of these things just do a quick Google search for them (this is the same process as for contributing to any open-source software on GitHub).

closed 2018-05-04 by ianpaschal

twosilly, 2018-06-01:
@ianpaschal Hello there! I'm trying to make my contribution to cura, but I found that I couldn't. My ability was limited, but I wanted to add some of my company's configuration files (printer definitions) to the great cura, I think I Can sponsor Cura's R&D, buy hard coffee for a cup of coffee or what I can do, I hope my printer model will appear in a cura version! Thank you (from an English idiot, please forgive me)