Ultimaker/Cura #3773: Printer settings do not save  nozzle size

karnhack, 2018-05-04:
**Application Version**
3.3.1

**Platform**
macOS High Sierra

**Printer**
Kossel Delta

**Steps to Reproduce**
When close and restart cura in printer settings  the nozzle size is reset at 0.5 instead 0.4

**Actual Results**

**Expected results**

**Additional Information**

diegopradogesto, 2018-05-04:
Which printer are you using? We support **Kossel Mini** and **Kossel Pro**. 
I checked both:
Kossel Mini is created with 0.4mm nozzle by default -> When I restart, the value is still there.
Kossel Pro is created with 0.35mm nozzle by default -> When I restart, the value is still there.

I can't reproduce your error. Please attach a project file (File>Save Project) so we can import it and check what happen.

fieldOfView, 2018-05-04:
I think the issue is when you change the 0.35mm nozzle of the Kossel Pro to 0.4. If I am not mistaken, @Ghostkeeper tried to fix this issue in Cura 3.3 beta, but that caused the issue that values would not "take" unless typed slowly, so he reverted his fix.

karnhack, 2018-05-04:
I was not looking for support I just wanted to report a bug... My configuration starts from DeltaBot option

ianpaschal, 2018-05-07:
Thank for taking the time to report.

The truth is though that reporting a bug in itself is not very useful. A bit like going to the police station to report a crime and refusing to give a statement.

In order to figure out what bug it is and how to fix it, we need to recreate it on our computers, and for that we need your project file as @diegopradogesto said.

diegopradogesto, 2018-05-07:
Don't worry @ianpaschal, I could reproduce it now. So I will create a ticket out of this issue. Thanks @karnhack for reporting, I was just looking for "Kossel Delta" and I couldn't find it with this name, so I was asking.

fieldOfView, 2018-05-07:
Note that this is a duplicate of https://github.com/Ultimaker/Cura/issues/3338 (which was fixed and then unfixed by @Ghostkeeper because the fix broke other stuff), so a ticket probably already exists.

diegopradogesto, 2018-05-07:
The other ticket was already done, so I've created a new one: CURA-5334

josteinaj, 2020-04-10:
This happens for me both in Cura 4.4 and 4.5 for my custom machine definition (ZMorph). I keep setting the nozzle size to 0.3mm, but I don't think it's saved, since it is back to 0.4mm when I reopen the nozzle settings. Any workaround for this?

bgutsev, 2020-04-15:
As a workaround, I found that there is a line which have to be deleted - it overrides the settings constantly.
C:\Program Files\Ultimaker Cura 4.5\resources\extruders\\<used_profile>.json
for example:
in C:\Program Files\Ultimaker Cura 4.5\resources\extruders\anet3d_extruder_0.def.json
I have deleted the line `        "machine_nozzle_size": { "default_value": 0.6 },`

The whole file content:
```
{
    "version": 2,
    "name": "Extruder 1",
    "inherits": "fdmextruder",
    "metadata": {
        "machine": "anet3d",
        "position": "0"
    },

    "overrides": {
        "extruder_nr": { "default_value": 0 },
        "machine_nozzle_size": { "default_value": 0.6 },
        "material_diameter": { "default_value": 1.75 }
    }
}
```
Hope it helps!

josteinaj, 2020-04-15:
@bgutsev thanks! I'll try

Ghostkeeper, 2020-04-21:
Yes we've had this regress, in a way. Try using a comma instead of a period. It depends on your OS language settings.

iglotov, 2020-05-05:
> Yes we've had this regress, in a way. Try using a comma instead of a period. It depends on your OS language settings.

Thanks, it works. It's replacing comma with dot again, but keeps new value as is should.