Ultimaker/Cura #3775: Post processing filament change, changes filament at the wrong layer

CPS-3, 2018-05-05:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

dmsulliva, 2018-05-06:
Same thing happens for me with my Tarantula, it was supposed to stop at layer 12, but stopped at 9 instead

diegopradogesto, 2018-05-07:
Yes, confirmed. I set the filament to change in layer 50 and instead the script set the change in layer 47:

`G0 X46.039 Y47.619`
`G0 X44.487 Y44.487`
`;TIME_ELAPSED:902.097665`
`;LAYER:47`
`G0 X44.487 Y44.487 Z4.8`
`M600 E30.00 L300.00 ; Generated by FilamentChange plugin`
`;TYPE:FILL`
`G1 F3600 X46.024 Y46.024 E149.4953`
`G1 X46.103 Y46.103`
`G0 F7200 X46.176 Y47.606`

diegopradogesto, 2018-05-07:
For developers' reference: CURA-5335

CPS-3, 2018-05-07:
@diegopradogesto thank you for confirming

djunisol, 2018-05-08:
yup same as CPS-3

southsidebrewer, 2018-05-12:
I have this exact same issue. I have to manually set any filament changes now. I am pretty sure this is a new issue with the latest release, but I do not know for sure.

AbeFM, 2018-05-13:
Every time I use this plug in I did test print columns to figure out what really happens.

Also the retract sometimes tries to *feed* several mm of material through nozzle at high speed.

Every single wrong in the macro meds to be tested each time to use safely. By adding or subtracting an offset, negating inputs, etc, it works.

CPS-3, 2018-05-28:
Is anybody working on this yet?

diegopradogesto, 2018-05-29:
It is in the planning.

CPS-3, 2018-05-29:
Ok thank you for the heads up

jackha, 2018-06-14:
Fixed in master. Unfortunately it just didn't make it to 3.4, so it's in the release after.

closed 2018-06-14 by jackha

diegopradogesto, 2018-06-14:
I thought that it is not dangerous and I just add it to the next 3.4 RC.

uwezi, 2019-03-05:
is the problem back in 3.6.0? It appears that the pause is inserted randomly at +/- 1 layers from the desired position

`;TIME_ELAPSED:1440.249614
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 27
M83
G1 F1500 E-15
G1 F300 Z6.4
G1 F9000 X190 Y190
G1 F300 Z15
M104 S150; standby temperature
M0;Do the actual pause
M109 S210; resume temperature
G1 F1500 E15
G1 F199.998 E3
G1 F1500 E-15
G1 F300 Z6.4
G1 F9000 X96.636 Y86.693
G1 F1500 E15
G1 F9000
M82
G92 E1608.46202
;LAYER:26
;MESH:Uwe Zimmermann.stl
G0 X96.636 Y86.693 Z5.4
;TYPE:WALL-OUTER
G1 F2100 E1608.46202`

Ghostkeeper, 2019-03-06:
Note that the `;LAYER` comment in the g-code is zero-indexed while Cura's interface shows the layer indices as one-indexed. So you'd expect Cura to indicate the layer number as one higher than what it says in the g-code.

abrenoch, 2021-02-13:
How is this still a problem?

[image]

[image]

With cura 4.8.0

AbeFM, 2021-03-16:
The process for change-at-z is slice-check-iterate. It should be documented.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes