Ultimaker/Cura #3776: Rotate, Pan, Zoom Controls - Gimbal Lock and Zoom Issue

Aeroengineer1, 2018-05-05:
Basics
Windows 10-64 Pro
Application version 2.0-3.3 (It may go back further, but this is as far back as I can remember)

The Issue
Overall, the controls for rotate, pan, zoom are rather challenging.  The main issue is that when rotating the view looking down along the z axis gimbal lock occurs.  This makes it challenging to inspect the model.  

Similar, when zooming, the zoom point cannot be altered.  When I select the model, it will center on the model, but after rotating, it does not always maintain the the zoom at the model center, nor does it maintain a zoom to cursor methodology found in other 3D modeling applications.

There are a few different systems that are used in CAD applications for avoiding these issues.  The two that I am most familiar with are CATIA and SolidWorks.  My preference is towards SolidWorks style.

I admit that I do not know the amount of effort to work an issue like this, but it would improve greatly the ability to inspect complex models after slicing.

ianpaschal, 2018-05-07:
Thanks for bringing this up. There's been a few other threads about this in the past which if I can find back I will link to this one. This is something that is on our radar but will unfortunately always receive lower priority than software-breaking bugs and crashes.

ianpaschal, 2018-05-07:
I'm going to close this in favor of trying to keep the discussion in one place. Feel free to repost your comment in... mmm... I think #3407 is most relevant.

Also related:
#3431, #3430, #3169, #2552

closed 2018-05-07 by ianpaschal

Ghostkeeper, 2018-06-06:
> The main issue is that when rotating the view looking down along the z axis gimbal lock occurs. This makes it challenging to inspect the model.

We decided on this being the lesser evil. Most CAD software has no gimbal lock because to them it doesn't matter that you're looking at the object upside-down. To a slicer this does matter though because you have a build plate. There is a definite bottom side and you always want that to be on the bottom. If you want to keep the build plate at the bottom you either have a gimbal lock or you rotate the camera 180 degrees when crossing the Z axis, which feels jolty.

> nor does it maintain a zoom to cursor methodology found in other 3D modeling applications.

There is a preference for this, "Zoom toward mouse direction".