Ultimaker/Cura #3778: Cura Can't Start

RanHam335, 2018-05-05:
I've never used Cura before. I installed it on my C drive and every time I open it I get a "Cura can't start" Dialog box. If I click show Configuration Folder the dialog box just closes out. Here is the detailed crash report:

System Information:
Cura Version: 3.3.1
Platform: Windows-7-6.1.7601-SP1
Qt Version: 5.8.0
PyQt version: 5.8.2
OpenGL: Not yet initialized

Error Traceback:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "Console.py", line 24, in <module>
  File "X:\3.3\build\inst\bin\cura_app.py", line 140, in <module>
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Application.py", line 340, in getInstance
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 242, in __init__
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 76, in __init__
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Application.py", line 94, in __init__
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Resources.py", line 67, in getPath
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Resources.py", line 119, in getStoragePath
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Resources.py", line 156, in getStoragePathForType
  File "X:\3.3\build\inst\lib\python3.5\site-packages\UM\Resources.py", line 363, in __initializeStoragePaths
  File "ntpath.py", line 113, in join
  File "genericpath.py", line 143, in _check_arg_types
TypeError: join() argument must be str or bytes, not 'NoneType'

Logs:
Current thread 0x00001348 (most recent call first):
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 364 in _logInfoWidget
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 214 in _createDialog
  File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 82 in __init__
  File "X:\3.3\build\inst\bin\cura_app.py", line 113 in exceptHook

ianpaschal, 2018-05-07:
First thing that stands out to me as a likely culprit is that you've installed to drive `X` instead of the typical `C`. Any chance you can try installing to `C` just to verify that that is indeed the issue? (Don't worry, we don't want to make that kind of requirement for our users, but it would help to verify before chasing this line of reasoning.)

fieldOfView, 2018-05-07:
> First thing that stands out to me as a likely culprit is that you've installed to drive X:\

The X: in the logs is a build system artifact, and does not mean that Cura was installed on that drive.

fieldOfView, 2018-05-07:
See the offending line:
https://github.com/Ultimaker/Uranium/blob/3.3/UM/Resources.py#L363

It seems like Resources._getConfigStorageRootPath() returns None. That means that on the @RanHam335's system, the APPDATA environment variable is not set. This is a variable that is set by default on Windows, and I would think that other applications would also have a problem with this variable not being set.

@RanHam335, to test this theory, could you open a cmd box and type the following:
```echo %APPDATA%```

ianpaschal, 2018-05-07:
Ah didn't know that about X:. But my eye jumped to the path because Resources._getConfigStorageRootPath() returns None.

jackha, 2018-06-06:
No reply in a month, let's close it.

closed 2018-06-06 by jackha