Ultimaker/Cura #3780: Cura 3.3.1 Exits to the Desktop When Refreshing Printer List

Intoxinator, 2018-05-06:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Ultimaker 3

**Steps to Reproduce**
1. Start Cura
2. Load model
3. Select Printer dropdown > Manage Printers
4. Click Refresh

**Actual Results**
Cura exits to the desktop

**Expected results**
List of Network Printers refreshed

**Additional Information**

smartavionics, 2018-05-06:
Where is the Refresh button?

[image]

Intoxinator, 2018-05-06:
Apologies - missed a step

Connect via Network > Refresh

[image]

smartavionics, 2018-05-06:
Sorry, I don't have a UM3 so I was unaware of that dialog. Well, I just tried pressing refresh on a Linux Cura and it didn't blow up - not much help to you though.

Could you find the cura.log and attach it please?

Intoxinator, 2018-05-06:
Found it! Stashed away good and proper. Cleared the logs, restarted and followed the steps. Critical error at the bottom.

2018-05-06 20:30:11,677 - DEBUG - [MainThread] UM3NetworkPrinting.DiscoverUM3Action.reset [47]: Reset the list of found devices.
2018-05-06 20:30:13,329 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-05-06 20:30:13,331 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-05-06 20:30:13,332 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Program Files\Ultimaker Cura 3.3\plugins\UM3NetworkPrinting\DiscoverUM3Action.py", line 61, in restartDiscovery
2018-05-06 20:30:13,333 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._network_plugin.startDiscovery()
2018-05-06 20:30:13,335 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "C:\Program Files\Ultimaker Cura 3.3\plugins\UM3NetworkPrinting\UM3OutputDevicePlugin.py", line 88, in startDiscovery
2018-05-06 20:30:13,336 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:     self._zero_conf = Zeroconf()
2018-05-06 20:30:13,337 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "zeroconf.py", line 1523, in __init__
2018-05-06 20:30:13,338 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: OSError: [WinError 10042] An unknown, invalid, or unsupported option or level was specified in a getsockopt or setsockopt call

[cura.zip](https://github.com/Ultimaker/Cura/files/1977611/cura.zip)

Intoxinator, 2018-05-06:
Python 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:04:45)

installed currently if that helps.

Intoxinator, 2018-05-06:
Of interest, found this thread:

https://community.ultimaker.com/topic/21185-cura-connect-to-um3e/

The OP seems to have exactly the same problem as me, but the thread got hijacked into a Mac related issue and there was no solution offered for Windows.

ChrisTerBeke, 2018-05-06:
Seems like an error in the underlying zeroconf library. There might be some device in your network that exposes itself on a port similar to the UM3 which causes the zeroconf to crash? Did you have issues before or when running Cura on another network?

Intoxinator, 2018-05-06:
I have had this issue for some time and switched to a Mac Mini for sending my jobs to the printer because of it. Have done some "reorganizing" and tried again to send jobs from my Windows laptop.

MSDN lists the error as:

WSAENOPROTOOPT (10042)
Translation: Bad protocol option.
Description: An unknown, invalid, or unsupported option or level was specified in a getsockopt (Wsapiref_8qcy.asp) call or a setsockopt (Wsapiref_94aa.asp) call.

https://support.microsoft.com/en-us/help/819124/windows-sockets-error-codes-values-and-meanings

Suggests that Zeroconf is doing something funky in its constructor (if that's what its called in Python land).

I don't know what port the UM3 is exposing, so can't answer that question, but guessing that we'd have conflicting errors on Mac if that were the case.

It might be worth spinning up a test harness that does what the underlying plugin is trying to do and see if we get any more meaningful error messages out of it.

I note that in my config for the printer, there's a manual entry containing the IP address of my UM3, but has a blank entry before it. The plugin enumerates those manual entries and splits on commas, so maybe it's trying to do a lookup on an empty IP address or something. It's a bit early and my Python isn't great.

:)

fieldOfView, 2018-05-07:
Something strange is that by the time you press the "Refresh" button, the zeroconf constructor has been executed a couple of times already (when starting the application and when opening the dialog). And obviously there were no winsock issues then. 

As far as I remember, pressing "Refresh" causes the zeroconf object to be (somewhat forcibly) destroyed and constructed again.

jackha, 2018-06-06:
Hi @Intoxinator can you try some of these things? See if we can figure out if it's computer related, or network related (or both).
- Disconnect network, see if Refresh still crashes
- Computer on a different network (see if it crashes)
- A different computer
- Different computer on a different network

Ghostkeeper, 2018-06-06:
Maybe Zeroconf has some static data that doesn't get reset when we re-initialize the zeroconf constructor?

Dredgon, 2018-09-07:
I have the same problem.
[cura.log](https://github.com/Ultimaker/Cura/files/2360448/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/2360449/stderr.log)

rburema, 2018-11-09:
Hi @Dredgon,
It sort of looks that way, but yours is sourced to a Repetier plugin instead of UM code, so maybe you should report it there ( https://github.com/VMaxx/RepetierPlugin ).
cheers, Remco.