Ultimaker/Cura #3781: [3.3.1] Incorrect/confusing color scheme legend in layer view

rickyzhang82, 2018-05-06:
**Application Version**

3.3.1
**Platform**

Mac/Linux

**Printer**

MP select mini
**Steps to Reproduce**

Open layer view and check the feed rate.

**Actual Results**

As you can see below, [image]

The color legend is very confusing regarding to the bright green vs dark green. You have no clue of feed rate due to the fact that both dark green show in two sides of bright green in the legend.

**Expected results**

I'd expect that bright green is more red shift than the dark green (if you follow the physic rule). As the wave length legend show below, 

[image]

Please change the color legend.

Also, I doubt the color is right. Because the outer wall should print slower than inside. Thus, it should use bright green in the inner wall and dark green in the outer wall in the first diagram.
**Additional Information**

rickyzhang82, 2018-05-06:
I'd propose the following:
1. I think the color it picked in the diagram needs to be in relative scale, instead of absolute scale for visualization purpose. Dark green vs bright green is harder to interpret than blue vs green. If you only show 5 different speed, you don't need a continuous spectrum and you can tweak it well enough with distinguishable color with discrete color legend. For my eyes, I'd think the inner wall is more bluish than outer wall. Thus, I thought that the print speed is faster in outer wall than inner wall. This is wrong according to the settings. See the diagram below: [image]
1. If picking color is hard to fix, change the color legend first. Use the visible spectrum above is way much better than what we have now.

ianpaschal, 2018-05-07:
I agree, it could be better. The really quick version in my mind is would be a blue-to-red gradient or something.

ChrisTerBeke, 2018-05-28:
Actually we specifically avoided red because it usually indicates warning/danger and we didn't want users to think parts of the print were not going to be printed correctly...

The current spectrums were chosen after quite some consideration with several people within Ultimaker.

rickyzhang82, 2018-05-29:
In layer view, inset is red line. The red line rule needs to be consistent.

[image]

ChrisTerBeke, 2018-05-29:
Sure, no need to get upset....

rickyzhang82, 2018-05-29:
Inconsistent rule is a big no no in any logic system.

Back to the fix, ditching the color spectrum should be easiest fix. Just pick several distinguishable color in advance and show the exact print speed with color as a dynamic legend. This alone can solve the problem with minimal code change.

Unless you show a heat map which contains tons of variable difference in the data, the color spectrum is an overkill here that leads to sub par usabilities.

ChrisTerBeke, 2018-05-29:
> This alone can solve the problem with minimal code change.

In that case, feel free to make a pull request to fix this :)

rickyzhang82, 2018-05-29:
Well, I'm working on 15.04 legacy branch for [an more important feature](https://github.com/Ultimaker/CuraEngine/pull/782) on my own. 

I live in US and wake up at 4AM to get this feature done asap. I don't have bandwidth to get this UI done.  In addition, I'm a back end developer. Solving UI is never my interest.

:-)

ChrisTerBeke, 2018-05-29:
Fair enough. We'll see if either the core team or someone from the community picks this up then.

Just a heads up, I don't think the legacy branch will ever be touched again by us, so that PR is probably not going to get merged. If you have ideas for improving the path planning for an up-to-date version of CuraEngine, we'd be happy to review and help out.

ianpaschal, 2018-05-29:
I can see how "red means under extrusion" might have made people avoid using it for the flow.

I think using 3 primary hues in the spectrum is a mistake, I'd use only two adjacent ones, like blue to green, or yellow to green, etc.

That being said, it's not _broken_ so it won't get any priority from us. It will have to come as a PR from someone.

printingotb, 2018-08-31:
Hi! Sorry for opening an old discussion.
If we temporarily set travel speed to something as low as the other settings, it becomes much easier to see the difference between different line types.
Perhaps a check-box whether to include travel moves in the preview or not would help?

printingotb, 2018-10-30:
Sorry, I meant to not include the "travel speed" for calculating the colour map.

closed 2026-09-06 by GregValiant