Ultimaker/Cura #3782: No retractions when first layer is printed

SorenBNielsen, 2018-05-06:
**Application Version**
Cura 3.3.1

**Platform**
All

**Printer**
FLXPrint233

**Steps to Reproduce**
Just start a print of a model and the first layer is devastating the print.

smartavionics, 2018-05-06:
It would be very helpful if you code post either the gcode (zip or change suffix to .txt) or even better the .3mf file (add .zip suffix). Thanks.

ianpaschal, 2018-05-07:
> The first layer is devastating the print.

Like, emotionally?

Please upload the project file in question (so we can diagnose not only the gcode but also the printer definition, etc.) otherwise there's not much we can do.

SorenBNielsen, 2018-05-07:
Ok, right, sorry for the lack of files. 
Here is a picture of the problem, where oozing is devastating the top layer of a display. 
[image]

I am attaching the project file and the G-code file here. 

[F_DisplayTop_180406C.curaproject.zip](https://github.com/Ultimaker/Cura/files/1979405/F_DisplayTop_180406C.curaproject.zip)
[F_DisplayTopx1_180406C_02_0912_3H.zip](https://github.com/Ultimaker/Cura/files/1979407/F_DisplayTopx1_180406C_02_0912_3H.zip)

SorenBNielsen, 2018-05-07:
Sorre, I think that I closed this issue, but it is still open.

smartavionics, 2018-05-07:
I tried loading your project but my Cura stopped because it doesn't have a definition for the printer you use.

SorenBNielsen, 2018-05-08:
I have attached the .stl file here, so you can verify that there is oozing in the first layer. 
[DisplayTop_180406C.zip](https://github.com/Ultimaker/Cura/files/1982839/DisplayTop_180406C.zip)
I have attached the Cura definitions for the printer, material and quality here. 
[CuraSetup_180508.zip](https://github.com/Ultimaker/Cura/files/1982875/CuraSetup_180508.zip)

smartavionics, 2018-05-08:
Sorry, those definitions are not compatible with my Cura version and I still cannot load your project - it's not your fault.

Looking at your gcode example, it seems to me that you do not have retractions enabled.

SorenBNielsen, 2018-05-08:
This is the settings used, so it seems to me that retractions are enabled: 
[image]

SorenBNielsen, 2018-05-08:
It is only in the first layer that the problem shows up. There are no problem in the next layers, where many retractions are performed.

smartavionics, 2018-05-08:
Set combing to "no skin".

SorenBNielsen, 2018-05-08:
THANK YOU ! That did the trick. I am attaching the first layer that is now printed perfectly. 
[image]

closed 2018-05-08 by SorenBNielsen

padcom, 2019-02-14:
Thanks for the info! My prints look a lot better now! Print longer (I don't really care about that) but the quality is just AWESOME by comparison!

Exi666, 2019-03-11:
thanks, I had the same problem too!

Jaggana69, 2023-12-25:
> Set combing to "no skin".

Thx a lot!!