Ultimaker/Cura #3783: UM2+ speeds don't match what the files say.

ChrisRiddell, 2018-05-07:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Ultimaker 2+ 
[Testing.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1978918/Testing.curaproject.3mf.zip)

**Steps to Reproduce**
Open cura make sure you are on a clean Ultimaker 2+ printer and check the print speeds

**Actual Results**
 Every materiel type you will see print speeds for infill at 100mm/s inner wall at 80mm/s but in the files on all quality types and in the Ultimaker2_plus.def.json you will see nothing overriding them speeds that high the default in the fdmprinter def is 60 and even in Ultimaker2_plus.def.json the override is `"value": "speed_print"`

**Expected results**
The print speeds to reflect what the definitions and quality files set. as they did in 3.3 I think the speed changes in 3.3 are a bug as 100 and 80 is just to fast as default in all my testing.

**Additional Information**
[image]

jackha, 2018-05-08:
Hi @ChrisRiddell , thanks for your report.
I think the difference between 80 and 100 greatly depends if the speed can actually be reached. And that depends on the model you're printing. For instance if you take speed 400 or 200, it will probably never make a difference in practice. However, if enable acceleration control and put higher acceleration in there, you would be able to make some difference.

alekseisasin, 2018-05-08:
Hi @ChrisRiddell , By default when you add a new UM2+ printer it will show right values. To get default values you can add a new printer and it will have already all default values.

jackha, 2018-05-08:
Ahhhhh yes I understand.

Ghostkeeper, 2018-06-08:
Did you solve this yesterday, @jackha?

Vandrasc, 2018-06-08:
Yes he did, it is currently in QA.

closed 2018-06-08 by Ghostkeeper

ChrisRiddell, 2019-05-09:
I think this is still an issue on the UM2+ ABS normal profile eg infill speed is calculated to 100mm/s when print speed is 55mm/s

Ghostkeeper, 2019-05-09:
No, that's correct. The `um2p_abs_0.4_fast.inst.cfg` profile specifies this:

https://github.com/Ultimaker/Cura/blob/602258bb4a1a08174c5fe55c346f18e5442be12b/resources/quality/ultimaker2_plus/um2p_abs_0.4_fast.inst.cfg#L29

We intend the infill speed to be 100mm/s. We make it a formula such that it ends up at 100mm/s if you use the default print speed of 55mm/s but it'll scale along with your print speed if you modify the print speed manually.