Ultimaker/Cura #3784: [3.3.1] Cura back to windows and crash when add printer

BugsBunny1403, 2018-05-07:
I have a custom printer settings work on cura 3.2.1 but with upgrade to 3.3.1.
The printer is an EXP3D, the builder provided configuration files for version 3.2.1 but I think with the changes in the machine manager a variable must be wrong and crash cura.

Cura.log said :
2018-05-07 11:07:15,390 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [210]: Starting to slice...
2018-05-07 11:07:15,390 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [227]: Going to slice build plate [0]!
2018-05-07 11:07:15,390 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [235]: Build plate 0 has no objects to be sliced, skipping
2018-05-07 11:07:23,937 - DEBUG - [MainThread] AutoSave.AutoSave._onTimeout [71]: Autosaving preferences, instances and profiles
2018-05-07 11:07:35,758 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [53]: Loading container exp3d
2018-05-07 11:07:36,080 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [exp3d] added.
2018-05-07 11:07:36,099 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [Exp3D_settings] added.
2018-05-07 11:07:36,100 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [Exp3D_user] added.
2018-05-07 11:07:36,127 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [fdmextruder #4_settings] added.
2018-05-07 11:07:36,129 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [fdmextruder #4_user] added.
2018-05-07 11:07:36,130 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[fdmextruder #4] added to [Exp3D] at position [0]
2018-05-07 11:07:36,136 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [fdmextruder #4] has already been added to this stack [Exp3D]
2018-05-07 11:07:36,143 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [355]: Container [fdmextruder #4] added.
2018-05-07 11:07:36,153 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [66]: An uncaught error has occurred!
2018-05-07 11:07:36,155 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: Traceback (most recent call last):
2018-05-07 11:07:36,156 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 381, in addMachine
2018-05-07 11:07:36,158 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]:   File "X:\3.3\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 109, in createMachine
2018-05-07 11:07:36,159 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [69]: AttributeError: 'NoneType' object has no attribute 'node_for_global'

diegopradogesto, 2018-05-08:
Can you provide the 3.2 data folder so we can investigate?

fieldOfView, 2018-05-08:
Also provide the configuration files for the printer (is they are not included in the 3.2 data folder)

BugsBunny1403, 2018-05-23:
I ve zipped the file of configuration for EXP3d

[Fichiers pour install manuelle.zip](https://github.com/Ultimaker/Cura/files/2030494/Fichiers.pour.install.manuelle.zip)

BugsBunny1403, 2018-06-04:
Up i always search error in config files with no succes

jackha, 2018-07-11:
This page will tell you where to look
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

Ghostkeeper, 2018-08-17:
I loaded up those profiles and got the same error. The mistake comes from the definition file. It names a `preferred_quality` metadata entry, but that should be `preferred_quality_type` in the more recent Cura versions. As a result, the preferred quality type remained `normal` and this quality type is not available for that printer.

I've made a fix so that Cura no longer crashes on this for the next release. You'll need to update the version of those configuration files anyway though. Good luck.

closed 2018-08-17 by Ghostkeeper

BugsBunny1403, 2018-08-27:
I ve done your advice but not resolve problem for cura 3.4.1
2018-08-27 23:35:33,009 - INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [74]: Extruder[fdmextruder #3] added to [Exp3D] at position [0]
2018-08-27 23:35:33,016 - WARNING - [MainThread] cura.Settings.GlobalStack.addExtruder [70]: Extruder [fdmextruder #3] has already been added to this stack [Exp3D]
2018-08-27 23:35:33,024 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [369]: Container [fdmextruder #3] added.
2018-08-27 23:35:33,028 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-08-27 23:35:33,028 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-08-27 23:35:33,030 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 391, in addMachine
2018-08-27 23:35:33,031 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "X:\3.4\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 114, in createMachine
2018-08-27 23:35:33,032 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'node_for_global'

BugsBunny1403, 2018-08-29:
@Ghostkeeper. I redid the procedure described to try to solve the problem but it was unsuccessful.
I don't understand what wrong.

Ghostkeeper, 2018-10-26:
Could you try again with Cura 3.5? The fix I made should be included with that release.

commit 31e28311 (2018-08-17): Check first if preferred quality exists

Fixes #3784.