Ultimaker/Cura #3787: Cura crashes when loading project that uses unknown printer - could be handled better?

smartavionics, 2018-05-07:
**Application Version**

master branch

**Platform**

Linux

**Printer**

Kossel XL (but this isn't the printer that caused Cura to crash, see below).

**Steps to Reproduce**

Load project file that references a printer that is unknown to the current Cura installation.

**Actual Results**

Cura dies.

**Expected results**

Cura should display an error dialog and not die.

**Additional Information**

>>>>
2018-05-07 11:04:59,237 - WARNING - [MainThread] cura.Settings.CuraStackBuilder.createMachine [37]: Definition FLX233 was not found!
2018-05-07 11:04:59,243 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [65]: An uncaught error has occurred!
2018-05-07 11:04:59,245 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: Traceback (most recent call last):
2018-05-07 11:04:59,245 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 875, in event
2018-05-07 11:04:59,245 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     return super().event(event)
2018-05-07 11:04:59,245 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 354, in event
2018-05-07 11:04:59,246 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     event._function_event.call()
2018-05-07 11:04:59,246 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2018-05-07 11:04:59,246 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     self._function(*self._args, **self._kwargs)
2018-05-07 11:04:59,246 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/python3/dist-packages/cura/Utils/Threading.py", line 27, in _handle_call
2018-05-07 11:04:59,246 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     ico.result = func(*args, **kwargs)
2018-05-07 11:04:59,247 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 561, in read
2018-05-07 11:04:59,247 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     return self._read(file_name)
2018-05-07 11:04:59,247 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:   File "/opt/cura/lib/cura/plugins/3MFReader/ThreeMFWorkspaceReader.py", line 601, in _read
2018-05-07 11:04:59,247 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]:     extruder_stack_dict = global_stack.extruders
2018-05-07 11:04:59,247 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [68]: AttributeError: 'NoneType' object has no attribute 'extruders'
>>>>

smartavionics, 2018-05-07:
This was the project I tried to load...

[F_DisplayTop_180406C.curaproject.zip](https://github.com/Ultimaker/Cura/files/1979697/F_DisplayTop_180406C.curaproject.zip)

fieldOfView, 2018-05-07:
This also means cura crashes like this if a definition fails to deserialise.

smartavionics, 2018-05-07:
That's true because I've seen crashes when the master branch has some change which affects the deserialization and if I fail to clear .cache/cura/... then it blows similarly.

diegopradogesto, 2018-05-08:
Confirmed: CURA-5337

diegopradogesto, 2018-06-11:
This is now fixed in master. Cura shows a message like this:
[image]

closed 2018-06-11 by diegopradogesto

smartavionics, 2018-06-11:
That's great, thanks!