Ultimaker/Cura #3791: [3.2.1] Option Mold (Special Modes) works only for extruder 1

tthurow, 2018-05-08:
**Application Version**
Cura 3.3.1

**Platform**
Windows 10 x64

**Printer**
Ultimaker 3

**Steps to Reproduce**

1. Load a model and select Special Modes -> Mold. CURA generate a mold model.

[image]

2. Selelct material from extruder 2 for the model. CURA generate a normal model. 

[image]

**Actual Results**
Only by extruder 1 option mold works.

**Expected results**
Option mold should be work on both extruders.

Vandrasc, 2018-05-09:
Hi @tthurow, silly question, but did you enabled the mold option for the second extruder as well? As it is not a global setting it has to be manually enabled in the second extruder tab as well.

Ghostkeeper, 2018-06-10:
I think that was the problem. I can't reproduce this bug in any case.

closed 2018-06-10 by Ghostkeeper

tthurow, 2018-06-10:
Oh no, sorry for so long time ... yes, I repeat this and it was my mistake! So sorry!