Ultimaker/Cura #3793: 3.3.1 - "Brim" is affecting "Optimize Wall Printing Order"

amagr0, 2018-05-09:
**Application Version** - CURA 3.3.1
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Select Brim, generate the gcode and in Layer View analyze the toolpath.

**Actual Results** - "Optimize Wall Printing Order" is not working.

With "Skirt" and "Raft" this situation does not occour.

BRIM (one wall at a time)
[image]

SKIRT (two walls)
[image]

RAFT (two walls)
[image]

[anprino_base_16x24.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1987277/anprino_base_16x24.curaproject.3mf.zip)

smartavionics, 2018-05-09:
Hi, thanks for the great report - this regression has already been fixed (see https://github.com/Ultimaker/CuraEngine/pull/750) for the next major release (3.4, I expect). Sorry for the inconvenience.

BagelOrb, 2018-05-15:
[Cleaning up fixed issue.]