Ultimaker/Cura #3794: Cura 3.3.1 Crash when I save gcode file

SMJ33160, 2018-05-09:
**Application Version** 3.3.1

**Platform** windows 10 pro / Intel Celeron N3450 8 Go SSd 64Go

**Printer** ANYCUBIC i3MEGA

**Steps to Reproduce** saving Gcode file

**Actual Results** Crash

**Expected results** slice done, I want to save the gcode file. Windows opens the windows file , I select the directory and give a name. I click Save and Cura crash without anything in the log . I install CURA in administrator mode or not it's the same issue.

[cura.log](https://github.com/Ultimaker/Cura/files/1987916/cura.log)

**Additional Information**
log issue from windows event (in french sorry ;-)

Nom de l’application défaillante Cura.exe, version : 0.0.0.0, horodatage : 0x5a2fae1f
Nom du module défaillant : _ssl.pyd, version : 3.5.2150.1013, horodatage : 0x5a2fa920
Code d’exception : 0xc0000005
Décalage d’erreur : 0x00000000000073e3
ID du processus défaillant : 0x2724
Heure de début de l’application défaillante : 0x01d3e179e2ba42da
Chemin d’accès de l’application défaillante : C:\Program Files\Ultimaker Cura 3.3\Cura.exe
Chemin d’accès du module défaillant: C:\Program Files\Ultimaker Cura 3.3\_ssl.pyd
ID de rapport : 5e961b0f-81d0-4aa7-b168-db772ecced62
Nom complet du package défaillant : 
ID de l’application relative au package défaillant :

SMJ33160, 2018-05-11:
Hi,

Sorry to be so late...

you'll find my user directory in the zip file

Thanks,
[cura.zip](https://github.com/Ultimaker/Cura/files/1995688/cura.zip)

cul8rmom1, 2018-05-18:
Mine crashes as well.  I can seem to get it to crash a lot of the time by putting the model under the build plate.  I.e. reducing the z value below 0.  When I am dont printing I will find a thingaverse model that causes it and let yall know.

Ghostkeeper, 2018-06-10:
It doesn't crash when I try it with your cura.zip folder :confused:

I tried:
* Loading a cube, saving to g-code.
* Saving to g-code under a file name that already existed.
* Saving to project file.
* Moving the cube to z = -6, saving to g-code.

Seems to work fine. I'm testing this on Cura 3.3.1 on Windows 8.1.

jdmcguire, 2018-06-13:
Mine does the same thing, but my model is more complex. I can do it in the Beta version for Ubuntu 18.10 no problem. I only encounter this issue in Windows 10

Ghostkeeper, 2018-07-31:
I tried this again (on 3.3.1) with more complex models. It doesn't want to crash. We're going to need something more to figure out how to reproduce the issue. Does it still occur in more modern versions of Cura?

diegopradogesto, 2019-07-02:
Is this the same as https://github.com/Ultimaker/Cura/issues/4016? Could you please try the workaround?

preacher4gh, 2019-07-08:
yes, I think so, got this confirmed in another forum and switching off feedback solves this issue

closed 2020-06-19 by no-response[bot]