Ultimaker/Cura #3795: Dual extrusion profiles are not synced to extruders

fakecrusader, 2018-05-09:
**Application Version**
3.2.1

**Platform**
Windows 7

**Printer**
Custom with dual extrusion, separate nozzles and heaters.

**Steps to Reproduce**
I want to print a model using ABS on extruder 1 and support material on extruder 2.

1 - Load model. Use "custom" option for print setup.

2- Choose extruder 1 settings and select an existing ABS profile I wish to use. Enable support and select E2 for the extruder to create support.
3 - Choose extruder 2 settings and select an existing support material profile I wish to use. This is similar to the ABS one but not identical. 
4 - Switch back to extruder 1. Look at active profile and settings.

**Actual Results**
When both extruders have different profiles, switching between extruders does not change the associated settings in the UI, the most recently loaded profile remains on view. Consequently, it is practically impossible to make changes and save the profile updates - I have messed up profiles because of this now. It is also unclear if both profiles settings are actually going to be used by the slicer or if the currently "active" one is being used. 

**Expected results**
Once a profile has been selected for the active extruder it should remain bound to it so that the settings display updates according to the active extruder. It therefore follows that saved changes to the displayed profile will update the correct settings. 

**Additional Information**
I've tried assigning different materials per extruder - these do remain bound however they are merely for information as far as I can tell and mainly redundant? (the profile name is a better guide to material imo).
I can understand why profile selections may act this way - having fully separate controls per extruder is a bit much and probably causes a lot of conflicting print plans per extruder in the 3D space. However, it is not unreasonable to have separate controls for temp, speed, retraction, etc, which don't have anything to do with the 3D space.

ChrisTerBeke, 2018-05-09:
Is this still an issue for you in the newly released 3.3(.1)?

fakecrusader, 2018-05-09:
It is now different in 3.3.1, and not in a favourable direction for mixed materials.
In 3.2.1, there is a menu tree where profiles are assigned to the extruders, i.e. Settings->Extruder1->Profile->current selected profile is marked in a list.
In 3.3.1 this tree is now Settings->Material or Disable/Activate. Profile is now at Settings/Profile.
This is totally illogical, it is essentially saying that you can assign different materials to each extruder but both materials must have the same extrusion characteristics, which are defined in a global profile that assumes one material. Material characteristics such as temp, extrusion rate, speed, retract, etc. should be profiled under Material. Only parameters that are not affected by a difference in material should be in the "Profile". 
I'm not too happy as I've been getting really good results from 3.2.1 on my single extrusion machine and having spent a lot of time and money on a second dual machine it seems that Cura is getting really messed up in that area.

ChrisTerBeke, 2018-05-28:
Well, they are. Materials are assigned per extruder, not globally. Temperature is in a material profile. The other settings are in a quality profile which is also assigned per extruder (normal quality PLA is a different profile than normal quality PVA, even if the name is the same in the UI). We get fine results with dual extrusion, as the Ultimaker 3 and S5 are our primary target machines for developing Cura.

Can you share a project file with your setup so it can be checked by one of the developers?

fakecrusader, 2018-05-30:
Hi thanks for the reply. 
I found something a couple of days ago (in 3.2.1) that explains the confusion. Having done hundreds of single extrusion prints, I've got into the habit of saving a profile based on material rather than print quality and I'm probably not the only one who has. I should really be setting up the material parameters in material management and then changing material. However, this method is also quite confusing as sometimes something like the temperature will change immediately, not change or say it's calculated(???). I think I tried using this workflow in an earlier version but found it unreliable so went for profiles as the means of recording settings per material.
What I found, as mentioned above, is that the pertinent separations **are** held in the current profile but they aren't visible in summary form without digging down into the individual profile under Preferences/Profiles. I will then see something that makes much more sense to me than what is contained in the parameter input UI as it is condensed down. Ideally the input UI should have the same filtering - as an ex-developer myself I realise this probably isn't going to be easy given that there is already a level of filter-switching going on due to dependencies in selected options. 
Having readjusted my mindset, I will take another look at the latest version again.

Cheers

closed 2018-05-30 by fakecrusader