Ultimaker/Cura #3796: Printing temperature drops to under 180 celcius

nickzomp1, 2018-05-10:
**Application Version**
Ultimaker Cura 3.3.1

**Platform**
Windows 10 64-Bit

**Printer**
FLSUN Klossl Delta

**Steps to Reproduce**
Load STL file into Cura to generate GCODE file. Set printing temperature to 220 celcius. GCODE file is then loaded into Repetier-Host (which I connect my printer to) to print. 

**Actual Results**
Printing temperature starts out at 220 celcius then drops to about 170-180 celcius (no matter what I set printing temperature to). Also at these temperature the feeder motor stops at times, then restarts at random.

**Expected results**
Printer should print at ~220 Celcius and feeder motor should not stop throughout print

**Additional Information**
I've noticed this happening recently on my prints. The temperature will drop to under 180 Celcius (PLA printing) and the feeder motor will stop running. I think this is because the heating element is not hot enough to melt the PLA filament, thus the feeder motor stops, but am not sure. Has anyone else had this problem?

smartavionics, 2018-05-10:
Please attach a typical gcode file so that I can check the temperature changes. Thanks.

nickzomp1, 2018-05-10:
Here's the last two files I've run, both with temperature issues (uploaded as .txt files):

[KM_CaddyRack2.txt](https://github.com/Ultimaker/Cura/files/1993339/KM_CaddyRack2.txt)
[KM_Gray Slaad for Shapeways.txt](https://github.com/Ultimaker/Cura/files/1993340/KM_Gray.Slaad.for.Shapeways.txt)

smartavionics, 2018-05-10:
As there are no gcodes in your files that alter the temp after the initial setting it must be something that is causing the hotend to cool. I notice that you are using 6.5mm of retraction. I don't know what extruder and hot end you are using but on my own Kossel Mini that is using an e3d Titan extruder and v6 hotend I am only using 4mm of retraction and don't get any stringing. It could be that you are over-retracting which could do 2 things that would decrease the nozzle temp (1) it will pull relatively cold air into the nozzle and (2) it will pull the filament out of the hot zone into the cold zone for the duration of the travel move. Another bad effect of over-retraction is that it causes "zits" to appear in random places on the print. I think those are caused by the air that is sucked into the nozzle forming bubbles which get expelled some random time later. So, try using less retraction and see what happens. One other thing, maybe you need a higher wattage heater in your hot end? You could try slowing down the print speeds and if the temp stays high then it's probably the heater not giving enough heat to keep up the flow rate required to print at the speeds you are using. Hope this helps!

AbeFM, 2018-05-13:
Does the extruder strip filament, or just so turning at all? 

Odds are the stop is a firmware preventing of extrusion - some firmwares even report such to a serial monitor.

Perhaps your thermocouple is loose, not seeing the hot end temperature (filament wouldn't strip, it will ooze and may burn of printer gets too hot.).
Heater could be loose, too, in head, or it's connection/wiring could be bad.

I would check all that, poorly controlled heaters can cause fires.

Lastly, power supply as suggested above, and consider a silicone boot for the heater block a la e3D.

BagelOrb, 2018-05-15:
Make sure all printing temperature settings are as high as required.
There are several settinga concerning temperature. Please check all by entering "temp" in the custom setting search bar.

BagelOrb, 2018-05-15:
Oh wait. It's not in the gcode so it's not Cura which is at fault.

closed 2018-05-15 by BagelOrb