Ultimaker/Cura #3797: Print thumbnail image contains support blocker blocks!

smartavionics, 2018-05-10:
**Application Version**

3.3.1

**Platform**

Any

**Printer**

S5

**Steps to Reproduce**

Load model into Cura and add support blocks, send job to printer.

**Actual Results**

The support blocks are shown in the print thumbnail image.

**Expected results**

The support blocker blocks should not be in the image?

**Additional Information**

I am reporting this issue on behalf of another.

[image]

ianpaschal, 2018-05-10:
Oh dear.

Vandrasc, 2018-05-16:
Ehm, @ianpaschal see CURA-5019, according to the comment this behavior is acceptable.

smartavionics, 2018-05-16:
LOL, really? How can an image that shows the model encased in concrete be acceptable?

smartavionics, 2018-05-16:
BTW, I notice that infill modifier meshes are also rendered in the image which, again, isn't that useful.

fieldOfView, 2018-05-16:
The fix is very trivial in PreviewPass.py; non-printing meshes are already filtered, they should just not be rendered, or rendered transparently.

ianpaschal, 2018-05-16:
@Vandrasc I actually don't see any comments there which say it's acceptable. I see that it's "Fixed" which is apparently is not.

ianpaschal, 2018-05-26:
This is fixed so should be immediately visible in master and coming soon to 3.4 beta.

closed 2018-05-26 by ianpaschal