Ultimaker/Cura #3798: 3.3.1  Not allowing (manual) printer control after paused print.

azn0boi04, 2018-05-10:
**Application Version**
<3.3.1>

**Platform**
<windows 10>

**Printer**
<Printrbot Metal Plus>

**Steps to Reproduce**
<During a print i would want to abort, first i would "pause", lift Z away from print, then abort to home machine. >

**Actual Results**
<When I want to stop/abort the print, i want to lift the extruder away from the print before i home/abort. 
Instead after i pause the print i am NOT able lift or shift in any directions, i used to be able to in previous versions, like 3.2.1 works ok with this. 
when i was adjusting the initial Z offset, it was too low and i wanted to abort print and lift Z.  but I usually pause first, lift Z, then abort, since i was not able to lift Z after a Pause, it cause the nozzle to drag on print bed. :( >

**Expected results**
<I would like to be able to pause and lift Z before i abort/home extruder. Cura 3.2.1 allows this movement. >

**Additional Information**

BagelOrb, 2018-05-15:
This seems to me to be a firmware issue, rather than a Cura issue. 

Where is the pause and abort button?
I don't think it's in Cura.

Ghostkeeper, 2018-06-13:
I think he means the pause and abort buttons when printing via USB cable, especially since he mentioned that it did work in 3.2.

Ghostkeeper, 2018-06-13:
I can't reproduce it though using the current development state. Was it fixed in 3.4?

azn0boi04, 2018-06-13:
oh i didnt know 3.4 was out, ill try to see if it was fixed, 
Yes i meant pause/abort button while printing (via usb) ,... then hitting a +Z. so the extruder head doesnt drag if the print is not going good in the first layer.

while printing, ill typically pause, Lift Z and then abort, it didnt allow me to lift Z before abort to home pos. 
ill try 3.4 tonight,

Ghostkeeper, 2018-08-28:
If we're not hearing any further complaints, we're going to have to assume it was fixed in 3.3 or 3.4 then.

closed 2018-08-28 by Ghostkeeper