Ultimaker/Cura #3799: Cura 3.2.1 / 3.3.1 clash when I change to layer view using large(?) STL data.

joonghochoe, 2018-05-11:
**Application Version**

Cura 3.2.1 / 3.3.1

**Platform**

Windows 10 (Notebook : 4G, Desktop : 8G Memory)

**Printer**

Raise3D N2 Plus Dual / Ultimaker S5 / ...

**Steps to Reproduce**

STL source(Dual Color Labrador) : https://www.thingiverse.com/thing:2738228

I used dual color stl file for dual extruder.  1st nozzle is used for PLA, 2nd nozzle is used for ABS. At print setup, check "Generate Support" and "Enable Prime Tower". And Scale up to 200%. Merge two models.
After Slice is OK, I changed view mode from "Solid view" to "Layer view"

**Actual Results**

At Process Layer 99%, Windows system(mouse, explorer,...) very slow down for a few minutes. And when I use Notebook, cura is stopped. And some debugging dialog window opens and cura closes.

Or when I use desktop, layer view is OK, but Windows system, Cura(scale, move, mirror,...) are very slow.

**Expected results**

Layer view change done. Windows system is OK.

**Additional Information**

Slice is OK, so the slice engine seems to have no problems. But there seems to be a problem with the simulation view plugin or pyqt, opengl, numpy???
[cura.zip](https://github.com/Ultimaker/Cura/files/1993869/cura.zip)

ChrisTerBeke, 2018-05-18:
I don't see any errors in the logs, but it might simply be that your machine ran out of memory.

Ghostkeeper, 2018-06-27:
Try switching to compatibility mode for layer view:

* Open Preferences.
* Tick "Force layer view compatibility mode (restart required).
* Restart Cura and load your model again.

I suspect that your GPU is running out of memory.

To fix that sort of thing, we'd have to apply compression to the vertex data. That's not easy to implement though. Or we'd have to leave out the normal vectors if the memory is not large enough, which might be more feasible...

jackha, 2018-07-10:
When I look at the logs I first suspect you have to update your graphics card drivers. 

The graphics card reports it supports all kinds of stuff, but it also reports that it's OpenGL 4.0. It should be at least OpenGL4.1

    2018-05-02 17:29:49,242 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
    2018-05-02 17:29:49,245 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.0.0 - Build 10.18.10.4653
    2018-05-02 17:29:49,248 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   Intel
    2018-05-02 17:29:49,250 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: Intel(R) HD Graphics 4000

joonghochoe, 2018-07-11:
Thanks,
Just now, Cura works well in version 3.4.
My notebook HP Ultrabook is a 5 year old product. So lastest intel HD4000 driver version is 10.18.10.4885. And Cura is operated at 4.0 compatibility mode.

closed 2018-07-11 by jackha