Ultimaker/Cura #3803: Changes to Filament Switching Behavior

AbeFM, 2018-05-12:
I have a couple suggestions that I think would help prints when using Chimera style printers:

The observed behavior:
When the end of a color has been reached, the printer stops in place, issues a retract on the active tool, switches the tool, issues a command to warm up to the profile's* "standby" temperature", waits until it gets there. When this happens it effectively simultaneously un-retracts, moves the head to the priming tower or ooze shield.

This leaves the inactive nozzle over the part oozing and the newly active one melting a hole in the part, at best.

It would be better for the print if we could leave the nozzle over the center of the ooze tower, until the temperature is reached, then switch them where the ooze should wipe on the inner/outer wall of the tower, then continue to print the tower as normal.

I think it's as simple as doing the move command before the tool switch routine.

Sorry is this is a poor post, its my first!

*Same as material's displayed definition.

**Application Version**
3.3.1

**Platform**
N/A - win10

**Printer**
ChiMara10

**Steps to Reproduce**
Slice.

**Actual Results**
Described above

**Expected results**
Less buns, blobs, and strings

**Additional Information**
I can provide a project file if you feel it helps!

AbeFM, 2018-05-12:
I added G1 X<tower coord> Y<Tower coord> as the toolchange end script and it seems to have helped a lot.

Do you need video or something or does that logic and report stand alone?

AbeFM, 2018-05-14:
Under "Machine Settings -> Extruder(N) -> Extruder End G-code" I put:

> G1 X180 Y150

That's it. :-) It's been working, and aside from the occasional unneeded travel on very small layers, I have no real complaints aside from having to manually move it around for different parts/towers.

Does seem like something which could be automated, but this allows for testing.

Ghostkeeper, 2018-06-13:
This is automated/implemented but probably not filled in for your printer definition. There's six settings that are relevant to this:
* `machine_extruder_start_pos_abs`
* `machine_extruder_start_pos_x`
* `machine_extruder_start_pos_y`
* `machine_extruder_end_pos_abs`
* `machine_extruder_end_pos_x`
* `machine_extruder_end_pos_y`

If you set the end and start position to the location of your prime tower, and the `_abs` settings to `true`, then it behaves as you expect it to. We didn't do that by default because other dual-extrusion type printers don't need this and for those it was seen as an unnecessary move causing blips.

Ghostkeeper, 2018-06-13:
You can't fill this position in in the interface though. It might be too specific for something that can just as well be achieved with the start and end g-code of the extruders. What do you think?

jackha, 2018-06-20:
We've discussed it and decided not to work on it.

joaomamede, 2019-08-20:
Is there anyway to use the wipe prime  tower position value from the interface to add to the end of extruder gCode? 
something like

G1 X{wipe_tower_x} Y{wipe_tower_y}

Ghostkeeper, 2019-08-21:
Yes. Almost correct. Use this:

    G1 X{prime_tower_position_x} Y{prime_tower_position_y}

drelich, 2020-08-26:
> Yes. Almost correct. Use this:
> 
> ```
> G1 X{prime_tower_position_x} Y{prime_tower_position_y}
> ```

This code works very nicely BUT it still does not deal with the fact that the nozzel 'parks' on the printed object where it finished the layer, retracts the filament, then moves to the coordinates, loads the other filament and does the prime tower.

What I'd like to see is that it immediately lifts the nozzle away from the printed objects while doing a quick initial retract to minimise oozing (or do a wipe into the infill).

Ghostkeeper, 2020-08-28:
> What I'd like to see is that it immediately lifts the nozzle away from the printed objects while doing a quick initial retract to minimise oozing (or do a wipe into the infill).

I'd expect it to move into the infill (if there is any infill close by) and retract there. Is that not happening for you?

Cura currently never retracts while moving, except if you use the Retract Continue script. Marlin deals with accelerations a bit weirdly if you try that.