Ultimaker/Cura #3804: "Prime Tower Purge Volume" option incorrectly hiding

AbeFM, 2018-05-12:
I can't change Prime Tower Purge Volume without enabling "wipe nozzle after switch". Perhaps this happened because they are consecutive on the options list.

**Application Version**
3.3.1

**Platform**
win10

**Printer**
Custom Dual Ext

**Steps to Reproduce**
enable/disable "Wipe Nozzle after switch" option under Dual Extrusion

**Actual Results**
Prime Tower Purge Volume dissapears

**Expected results**
It shouldn't hide

diegopradogesto, 2018-05-13:
The "Prime Tower Purge Volume" is used when the "Wipe Nozzle After Switch" is enabled as indicated in this line: https://github.com/Ultimaker/Cura/blob/527f269b5d61614212f803c0321e84a9d70cd8bc/resources/definitions/fdmprinter.def.json#L5023

That's why the corresponding text field appears only when the ¨Wipe Nozzle After Switch" is selected. I don't see the problem here.

Are you sure this is the setting you want to play with? Let me know if I can help.

AbeFM, 2018-05-13:
Perhaps I'm using the feature in an unintended way.

The issue is towers of alternating materials can have poor layer bonding, their small contact area and under heated nozzles cause a weak layer which future layers stress and peel with increasing torque as tower gets taller.

I use the priming to try to get the material in the best state for sticking. I don't need to wipe as the next thing I touch is the disposable prime tower.

Basically I'm avoiding the wipe because the router already rarely works.

Lastly the priming is still done with wipe turned off. If the behavior was as described, shouldn't you turn off the prime when wipe is disabled?

AbeFM, 2018-05-14:
Thanks. I could swear yesterday that wasn't the behavior I saw, but it certainly is today.

Seems like this topic is done. Thanks again!

closed 2018-05-14 by AbeFM