Ultimaker/Cura #3805: Primary Nozzle Doesn't Preheat if Brim is with Secondary Nozzle

AbeFM, 2018-05-12:
When I set "build plate adhesion extruder" to Ext"2", the Ext1 does not get preheated, excess hard ooze can knock brim off build surface, Ext1 works poorly when it begins printing.

**Application Version**
3.3.1
**Platform**
win10
**Printer**
Dual Ext

BagelOrb, 2018-05-15:
What kind of dual extrusion printer do you have?
Did you select one from the list or did you make a custom dual extrusion printer?

Please attach your project file.

AbeFM, 2018-05-15:
I believe I took the CR-10 profile and edited it. It is quite likely there are some issues there.

[DualExtChiMara_Testing.curaproject.zip](https://github.com/Ultimaker/Cura/files/2005936/DualExtChiMara_Testing.curaproject.zip)

Please look it over and let me know. Everything works right if you brim with the other nozzle.

Ghostkeeper, 2018-06-13:
When slicing that project file they do both get pre-heated. It starts the g-code right off the bat with this:

    T1
    M190 S60
    M104 S205
    M104 T1 S192
    M109 S205
    M109 T1 S192

So I'm not able to reproduce the problem.

I would expect the second nozzle to get pre-heated to stand-by temperature and shortly before needing to print to get pre-heated to the starting temperature. In this case that doesn't apply since the stand-by temperature is the same as the starting temperature.

jackha, 2018-06-19:
@AbeFM is the problem still there? Can you verify if you also share @Ghostkeeper 's outcome?

AbeFM, 2018-06-19:
```
T1  Select Tool 1
M190 S60  Set Bed Temp 60
M104 S205 Set Active Tool (#1) to 205
M104 T1 S192 Set Tool #1 to 192
M109 S205   Set active tool (#1) to 205 and wait for temp
M109 T1 S192 Set tool #1 to 192
```

To my mind, this code doesn't touch Extruder 0 or its temp in any way.

I still get the priming of Ext0 not happening if the first printing is with Ext1 because the extruder isn't warm enough to allow extrusion during that prime.

If the first printing is with Ext0, everything works fine.

Ghostkeeper, 2018-07-31:
> To my mind, this code doesn't touch Extruder 0 or its temp in any way.

Er, indeed. I don't know why I didn't see that. I tried again on 3.4.0 and now I get this g-code:

    ;FLAVOR:Marlin
    ;TIME:50140
    ;Filament used: 47.9217m, 14.3546m
    ;Layer height: 0.26
    ;Generated with Cura_SteamEngine 3.4.0
    T0
    M190 S60
    M104 S208
    M104 T1 S192
    M109 S208
    M109 T1 S192
    M82 ;absolute extrusion mode
    G28 W ; home all axes without mesh bed leveling
    T0;
    G92 E0 ; reset extrusion distance
    T1; Switch Ext
    G1 X100.0 E12.5  F1000.0 ; priming
    G92 E0; Zero Ext 1
    G1 E-2.6000 F2700; Retract
    T0; Switch Ext
    G1 X60.0 E9.0  F1000.0 ; priming
    G92 E0; Zero Ext0
    G1 E-2.6000 F2700; Retract
    G92 E0
    G1 F2700 E-2.6
    ;LAYER_COUNT:775
    ;LAYER:0
    M107
    G0 F3600 X102.786 Y151.373 Z0.3
    ;TYPE:SKIRT
    G1 F2700 E0
    G1 F1650 X102.004 Y153.716 E0.12323
    G1 X102.175 Y154.565 E0.16644
    G1 X102.186 Y154.726 E0.17449

This does heat T0 as well as T1.

Mind that the priming seems to be done by the printer's start g-code for this ChiMara printer. If that start g-code sets its own temperatures, it is up to the maker of that profile to set it before priming. In this case it seems that the temperatures are not set by the start g-code, so Cura puts it in front of the start g-code.

Long story short, indeed I did reproduce the bug in 3.3, but not any more in 3.4. Are you seeing the same?

AbeFM, 2018-08-01:
Oh awesome. Looks like I porked the piggie on that one.

That is my start code.

Oh, and I added that *because* it wasn't preheating. It's all coming back
to me now.

How should I go about ensuring the nozzles are heated on time then?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

cicinovec, 2018-10-20:
Hello,
today tried with 3.5.1 and the issue remains. When brim is made with extruder 1 (T1), it does not preheat the T0 since the G-code tells it only to heat T1.
The issue is in line 5 (code M104 S205). Since Marlin has already chosen T1, M104 S205 is again related to T1.

The only solution so far for me is to change the brim to T0 material or adapt the G-code manually.

```
;FLAVOR:Marlin
;TIME:27696
;Filament used: 4.36283m, 0.8331m
;Layer height: 0.1
;Generated with Cura_SteamEngine 3.5.1
T1
M140 S80
M105
M190 S80
M104 S205
M104 T1 S200
M105
M109 S205
M105
M109 T1 S200
M82 ;absolute extrusion mode
G28 ;Home
G1 Z5.0 F300 ;Move the platform down 15mm
T0
G92 E0 ;zero the extruded length
G1 F200 E15 ;extrude 15mm of feed stock
G92 E0 ;zero the extruded length again
T1
G92 E0 ;zero the extruded length
G1 F200 E15 ;extrude 15mm of feed stock
G92 E0 ;zero the extruded length again
```

AbeFM, 2018-12-13:
I still have a slightly involved start code. I believe I had to do it, but I don't specifically remember what drove me to it.
Reading it, there's no preheating going on, so let me just slice something real quick.

Ok: He's right. It doesn't preheat the nozzle. Which I used to hate, but now I like.

Before I explain, let me note: It DOES preheat the nozzle some 300 lines of g-code before the toolchange command (with my settings) So it shouldn't cold extrude.

Plus side: If you're NOT using the second extruder for the first XX% of the print, it prevents oozing and such during a possibly significant percentage of the print.
More importantly, if you're not using something like PVA, it may not be wise to cook it all that time.

Minus side: technically, the standby temperature should take care of this and you could preheat that amount. And of course, when you go to use the nozzle, it won't have had the benefit of any purging. Something like a "first time use" (start code for the nozzle) might solve all this.
I'm not convinced it's an approach that would make it easier for the user.

What issues are you having as a result of this?

closed 2020-06-20 by no-response[bot]