Ultimaker/Cura #3806: Upper case .STL file extension doesn't work right on Linux

smartavionics, 2018-05-12:
**Application Version**

Master branch but a forum user reports problems with 3.3.1

**Platform**

Linux

**Printer**

Any

**Steps to Reproduce**

Load an stl file with an upper case file extension, e.g. simple_vase.STL

**Actual Results**

No print job name is displayed in bottom right corner and output file name is empty.

**Expected results**

The file name should be as it is when the extension is lower case, for my printer it would be KMMB_simple_vase.gcode

**Additional Information**

Cura log says this on loading file with .STL extension:

````
2018-05-12 21:44:24,774 - WARNING - [MainThread] cura.PrintInformation.setBaseName [346]: Unsupported Mime Type Database file extension
````

See also https://community.ultimaker.com/topic/22883-printer-name-prefix-on-g-code/

smartavionics, 2018-05-13:
Just checked 3.2.1 and that copes with .STL OK so this is (yet another) regression in 3.3.

Vandrasc, 2018-05-16:
Yes, we noticed as well, it works just fine on win though. @Ellecross FYI

Vandrasc, 2018-06-07:
The fix will be in the 3.4 RC, addressed in CURA-5367.

closed 2018-06-07 by Vandrasc