Ultimaker/Cura #3808: [3.3.1] Ultimaker Original Plus bed leveling broken

tevd, 2018-05-13:
**Application Version**
3.3.1

**Platform**
Windows 10

**Printer**
Ultimaker Original Plus

**Steps to Reproduce**
Start bed levelling

**Actual Results**
X, Y and Z axis home, but afterwards the Z axis goes down about 5mm.
This makes it impossible to level the print bed.

**Expected results**
X, Y and Z axis should home and stop moving.

**Additional Information**
It is not a hardware issue, since it worked perfectly with Cura 2.7.

ChrisTerBeke, 2018-05-18:
@nallath related to the networking rewrite?

Tschingis, 2018-05-31:
I have exactly the same problem in Version 3.3.1. Its also broken in Version 3.2.1.
Previously there was a button Move Z Up after Start bed leveling, this is gone.

alekseisasin, 2018-06-20:
@tevd Similar issue was already handled and the solution was not in 3.3.1. The issue was related to printer could not apply bed leveling in right way.
Soon will be released stable 3.4 Cura version and you can test there.

alekseisasin, 2018-06-28:
Ping @Tschingis , @tevd  
Did you try bed leveling in new Cura 3.4 version?

tevd, 2018-06-28:
I just tried it, and it works now

closed 2018-06-28 by tevd

alekseisasin, 2018-06-28:
@tevd , Thank you. If you will have other difficulties with Cura please let us know.