Ultimaker/Cura #3810: AttributeError"

cncpadawan, 2018-05-14:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

cncpadawan, 2018-05-14:
I should also note that in the configuration folder, the only 2 folders that get created are plugins and set_visibility, which are empty folders.

cncpadawan, 2018-05-14:
Another note - the code being pulled is still the current from the 3.3 branches for cura and curaengine.

smartavionics, 2018-05-14:
Hi, I also build from source and sometimes I have to also delete the cura cache folder. On Linux it is located at ~/.cache/cura but I don't know where it is on other systems, sorry.

cncpadawan, 2018-05-14:
Hello smartavionics, 

Thanks for the tip!  Hmmm..  The build server should be pulling and building everything clean (except the build environment) so in theory it should be clean every time.  Do you think this is something to do with the way it was built?.

smartavionics, 2018-05-14:
Hi, the cache I am referring to is used by cura when it runs not during the build. I don't know much about how the front-end python code works but I think it holds binary versions of the various profiles and configuration data and it sometimes gets out of sync with the text versions.

cncpadawan, 2018-05-14:
Ah, ok... I think you may be thinking of the folder created in AppData?  I did delete that, and I deleted all folders in Program Files related to Cura (it left some remnants). 

I just tried creating a Hyper-V VM and installing it in there.. that didn't go so well (it blew up, no OpenGL support), so I'll try a laptop that I have never installed Cura onto before and see what I get there.

smartavionics, 2018-05-14:
OK, good luck!

cncpadawan, 2018-05-14:
Thanks!  But... same issue.  Same errors.  Also, it seems the error dialog doesn't like 4k monitors that well... lol.

So it seems to be something in the way my system is building it.

cncpadawan, 2018-05-14:
Anyone have any other thoughts on this?

cncpadawan, 2018-05-18:
No other thoughts or ideas on this?  Nobody else is running into this when building?

cncpadawan, 2018-05-20:
I'm the only one seeing this, or has seen this?

diegopradogesto, 2018-05-22:
If you are running from source you should be able to do your own research on what is happening because it can be related with your own environment. If you go exactly to the point where the error points to you, it's line 753 of the file CuraApplication.py, in the function runWithGUI. The error says `AttributeError: 'NoneType' object has no attribute 'setOrigin'` that in this case means the `camera_tool` was not correctly initialized since it's still `None`. Then I encourage you to check in the logs whether the `CameraTool` plugin was loaded or not. You should see something like **`Loaded plugin CameraTool`**. If you see something different, then you need to start debugging from there.

cncpadawan, 2018-05-24:
Diegopradogesto, thanks for the info!  The problem is there ARE no logs created for me to see what loaded and what didn't load.  I understand it could have something to do with the build environment.. but since it has worked fine with 2.x, I'm trying to understand what would have changed, i.e. narrow the issue down.

cncpadawan, 2018-06-11:
Well, got it figured out.  The issue was the build environment.  Apparently there were some undocumented changes to the build environment requirements.  Rebuilt the build environments (for both Windows and Mac, both have exactly the same issue) and the problem is gone.

closed 2018-06-11 by cncpadawan

diegopradogesto, 2018-06-11:
Oh, good to here that you found the problem. You said that there are some undocumented changes in the requirements. Can you tell me which ones so we can update the documentation properly?

cncpadawan, 2018-06-12:
Unfortunately, I can't say for certain.  I rebuilt the build environment from the environment build github.  I am guessing something changed in there but I can't say for certain what it was.  The last time I built the build environments, it was in the 2.x era.