Ultimaker/Cura #3812: First save of project doesn't append .curaproject.3mf to suggested file name

smartavionics, 2018-05-14:
**Application Version**

3.3.1

**Platform**

Linux

**Printer**

Kossel Mini (or any other as far as I can tell)

**Steps to Reproduce**

Start cura afresh, load a model, fiddle with settings, etc. and then save as a project.

**Actual Results**

File save dialog appears with the model name but no .curaproject.3mf suffix

**Expected results**

File save dialog should include the .curaproject.3mf suffix.

**Additional Information**

[image]

Slicing and then saving to file does show the .gcode extension (i.e. KMMB_wolf.gcode in my above example).

Vandrasc, 2018-05-15:
Devs, see CURA-5368. 

Thank you, @smartavionics !

LipuFei, 2018-06-07:
@smartavionics Fixed in master: 3a7a1dbc6d935b94e4c9b080259618f2f55551bf

closed 2018-06-07 by LipuFei