Ultimaker/Cura #3815: Layer Not in correct Z height

Ritch2125, 2018-05-14:
For some strange reason Cura placed a layer that should around 24mm at z=0.  See attached screen shots

At layer 171 all is fine
[image]

layer 172, suddenly there is a new lines at the z=0 level

[image]

Finally at the next layer you can see the gap in the object
[image]

[TBW_Fan Mount  30mm Vertical - Fan Mount 30mm Vertical final.3mf.log](https://github.com/Ultimaker/Cura/files/2002942/TBW_Fan.Mount.30mm.Vertical.-.Fan.Mount.30mm.Vertical.final.3mf.log)

Version 3.3.1 on Windows 10.
Tevo BW

The 
[Fan Mount  30mm Vertical - Fan Mount 30mm Vertical9.stl.log](https://github.com/Ultimaker/Cura/files/2002953/Fan.Mount.30mm.Vertical.-.Fan.Mount.30mm.Vertical9.stl.log)

Any thoughts?

fieldOfView, 2018-05-15:
I wonder if it is actually in the gcode like that, or a visualisation glitch in Layer View. Does the layer stay out of place like that if you save the gcode and load it back in?

ChrisTerBeke, 2018-05-15:
Having the gcode would also help to debug this for the developers.

smartavionics, 2018-05-15:
Yes, please post the gcode for that. Thanks.

Ritch2125, 2018-05-15:
I didn't save the gcode.  But I will try and relod it tonight to get the
gcode.  I did try to switch layer heights to make the error go away but it
stayed.  Can the gcode be created from the FMF file?

BagelOrb, 2018-05-16:
[reminder] can you reproduce it?

Could you please share both the gcode and the project file? (Put them in a .zip)

Ritch2125, 2018-05-16:
Tim, I loaded the fmf fille already.  I think the stl as well.  They both
were remained with a .log on the end.  I did not save the gcode when I
spotted the error.  I should of. Time was tight last night and after a
couple of attempts I could not reproduce the error to get the code. I will
try again tonight.

Ritch2125, 2018-05-16:
I tried for 40 mins to recreate the problem but couldn't.  I will watch for this in the future and make sure I save the gcode next time.

smartavionics, 2018-05-16:
OK, thanks for trying. If it happens again, please save the gcode and attach to this issue. Cheers!

closed 2018-05-16 by BagelOrb

Ghostkeeper, 2018-06-15:
I've seen this problem once in a while too. I think it's some kind of data corruption inside the socket or a data race from CuraEngine. It happens very rarely, maybe once a month (with my 40 hour per week job). This bug has been in there for at least 2 years.