Ultimaker/Cura #3818: Feature Request - Brim only on Prime Tower

amagr0, 2018-05-15:
Problem: The low area contact of the Prime Tower cause tower unstick. 
Solution: The possibility to choose Brim only for the Prime Tower.

cmatthes, 2018-05-15:
I think it would also be useful to have an option for a multi layer brim. Creating an even better base.

cmatthes, 2018-05-15:
A raft would probably work too. 
I would choose a brim over a raft because a brim is way simpler for the user as it requires far less settings to be set. 

Enable Prime Tower Brim   [ X ]    
   - Brim width                     [     ]
   - Brim Thickness             [     ]
      - Brim Layers                [     ]

ChrisTerBeke, 2018-05-18:
Related to https://github.com/Ultimaker/Cura/issues/3824.

jackha, 2018-06-20:
Devs: see CURA-5487

printingotb, 2018-07-31:
I also think a brim on supports only will be really useful.
What would be even better is if we were also able to print a solid initial support layer against the buildplate.

AbeFM, 2018-08-01:
A raft only for supports is what I really want.

Dense layer, raft, brim, all these ideas point in one direction: support
adhesion is important to people who use soluble supports.

Perhaps on a case by case basis you can dismiss it, but ten topics that are
all variations on a theme says that this is something there is interest in.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

BagelOrb, 2019-10-29:
Cura 4.3 has Prime Tower Brim feature. This feature is implemented.
[image]

closed 2019-10-29 by BagelOrb