Ultimaker/Cura #3819: Moves from printer settings don't show in G-code preview

AbeFM, 2018-05-15:
[image]

**Application Version**
3.3.1

**Platform**
win10

**Printer**
Custom CR10/Dual Ext

**Steps to Reproduce**
"Extruder End G-Code" in machine settings set to:
`G1 X150 Y280`
on both tools. Slice File, Layer View

**Actual Results**
[image]

**Expected results**
Toolpath should show going beyond the tower?

**Additional Information**
Not terribly certain this is a bug.

BagelOrb, 2018-05-16:
Yes this is a bug. ...or a feature.

We never implemented interpreting the gcode. The layer view is from what Cura thinks it is doing and Cura doesn't look inside the user provided gcode settings.

BagelOrb, 2018-05-16:
However, it is not a problem for Ultimaker printers, so Ultimaker is not going to spend time on this.

fieldOfView, 2018-05-16:
> Cura doesn't look inside the user provided gcode settings.

The same is true for moves in the "global" start and end gcode

Ghostkeeper, 2018-06-13:
This issue is a duplicate of https://github.com/Ultimaker/Cura/issues/3128 .

You can still see the start and end g-code by saving the g-code to file and re-loading that g-code into Cura.

closed 2018-06-13 by Ghostkeeper