Ultimaker/Cura #3822: Cura adds M82 to the end of my gcode files.

amedsoprano, 2018-05-17:
at the en of my endscript cura always adds an `M82` to set the E to absolute extrusion, so when i go and print the next file it does crazy stuff. (yes, i print in relative)

smartavionics, 2018-05-17:
Yes, the Cura devs demanded that at the end of the print, it resets the absolute/relative extrusion mode back to "normal". They considered all gcode flavours except RepRap to normally use absolute extrusion. I said at the time that this was a mistake but they had their way. Sorry, it's a feature (intended bug). So the only way around this bug is to choose RepRap flavour gcode but that may not be compatible with your firmware.

fieldOfView, 2018-05-17:
If the only (?) problem is that the next print "does crazy stuff", couldn't you add M83 to the start gcode as a workaround?

smartavionics, 2018-05-17:
But what happens when the OP is also using gcode produced by other slicers that don't have this bizarre behaviour?

fieldOfView, 2018-05-17:
OP could also add the M83 there. I said its a workaround, not a fix.

fieldOfView, 2018-05-17:
A postprocessing script to remove the offending M82 could also work (could be as simple as using the built-in search/replace to replace any ```M82``` with ```;M82```). Still a workaround, not a fix, but at least postprocessing scripts now persist between sessions.

smartavionics, 2018-05-17:
Good thinking, that could be a usable workaround.

amedsoprano, 2018-05-17:
Yep I added M83 to the start script, but it sure is a workarround

jackha, 2018-06-20:
Depending on the setting "relative_extrusion" in the printer definition, Cura will put either M82 or M83 in the gcode. Hope this helps.

closed 2018-06-20 by jackha